trelau/pyOCCT #14: conda install -c trelau pyocct PackageNotFoundError

n4562, 2018-05-18:
```
conda create -n occt python=3.5
conda activate occt
conda install -c trelau pyocct
```

gives me a:

```
PackageNotFoundError: Package missing in current linux-64 channels: 
  - pyocct

Close matches found; did you mean one of these?

    pyocct: occt, pysot, pydot
```

looooo, 2018-05-18:
@trelau doesn't have linux-packages. you can try using
`conda create -n occt pyocct -c cad`

trelau, 2018-05-19:
@n4562 @looooo is right, my channel only has win64 builds at the moment. The Appveyor and Travis-CI builds work, but sometimes hit memory limits and fail, so I haven't been uploading the conda packages automatically. I'll see if I can find a successful one to upload. The dependencies should be available via conda, so you could try building it yourself. If you do, let me know if you have any problems with the build.

trelau, 2018-05-20:
@n4562 Linux packages should now be available. Please report an issue if you have problems. https://anaconda.org/trelau/pyocct

closed 2018-05-20 by trelau

HuaJianDeng, 2018-12-11:
@trelau Hello,

I am a beginner to python. I found your pyOCCT very helpful so I want to use it. But some problems occured when I followed the instructions to install.

conda create -n occt python=3.5
conda activate occt
conda install -c trelau pyocct

then, it showed:

1. Solving environment: failed

2.PackageNotFoundError: The following packages are not available from current channels:
  -pyocct
  -netgen ==6.2.1804
  -pyocct
  -occt ==7.2.0
  -pyocct
  -smesh == 8.3.0b0

Could you please offer me some suggestions ?  I will appreciate it very much.