trelau/pyOCCT #17: Bindig generator not open sourced

adam-urbanczyk, 2018-05-21:
You have a very interesting project here! We are considering reusing it to implement a new version of cadquery (cf. https://github.com/dcowden/cadquery/issues/268 ) to get the latest OCCT 7.2 capabilities and bugfixes.

One drawback that I see is that the code (I assume that there exists one) used to generate the pybind11 glue code is not share. This poses some risks w.r.t. long term support of the project. Do you maybe plan to share that code at some point of time?

trelau, 2018-05-21:
@adam-urbanczyk thanks for the interest. yes, there is a binding generator and yes i intend to make it available in the near future. the first iteration was a bit of an ad hoc approach to get the bindings done quickly to support the LaughlinResearch/AFEM (https://github.com/LaughlinResearch/AFEM) project. It actually worked quite well with only a relatively few handful of edge cases to manually tweak.

I'm doing a major overhaul of the tool before making it available to avoid confusion since there is still a lot of churn. Once it is stable the plan is to open source it.

adam-urbanczyk, 2018-05-21:
@trelau  done. Looking forward to the release!

trelau, 2018-06-27:
@adam-urbanczyk slowly making progress, but need some more time. it's a pretty large refactoring so I'm hoping to get things organized before releasing so help can focus on specific issues and edge cases. perhaps in the next week?

adam-urbanczyk, 2018-06-28:
Thanks, looking forward!

adam-urbanczyk, 2018-09-09:
@trelau  in the meantime I started working on my own generator using libclang and jinja2: https://github.com/CadQuery/cpp-py-bindgen Not yet fully functional, but feel free to share your thoughts!

trelau, 2018-09-10:
@adam-urbanczyk awesome. i'm still picking away at it. it's been tough to find time with the new job. it's not yet fully functional, but maybe i'll post it anyway in case someone can pick up the slack!

trelau, 2018-09-11:
open sourced https://github.com/LaughlinResearch/pyOCCT_binder any help welcome!

closed 2018-09-11 by trelau

adam-urbanczyk, 2018-09-11:
Great, thanks for sharing your code!