trelau/pyOCCT #18: Using PySide2 with pyocct

ffvpor, 2018-09-01:
I discovered pyocct few days ago and it matched perfectly with my own needs (pybind11 and last versions of opencascade). Thanks for that!

I use elementaryOS in a virtual machine and installed conda and pyocct. During my first tests with code examples, wxpython and his "RuntimeError: Xw_Window, given invalid X window" stopped me! I tried a solution from https://github.com/tpaviot/pythonocc-core/commit/5b9d1f468ef8a1e011e342d5b82fa4ffbf40a462 without success.

For my own projects, I adapted your old pyside code with pyside2 (after installing pyside2 5.6.0a1 py35h45ccf91_3 from conda-forge). It works but to see the compressor, I have to resize main window and keypress "fit" (Note: Apart from that, move, rotate and keypress are OK)

Console message: "TKOpenGl | Type: Other | ID: 0 | Severity: Medium | Message:
  OpenGl_Window::CreateWindow: window Visual is incomplete: no depth buffer, no stencil buffer"

Perhaps an opencascade bug (https://www.opencascade.com/content/samples-tutorial-gui-1-shows-no-icons-2-mnunew-reports-errors)?

For now, my lack of skills keeps me from resolve myself this issue, could you help me?

My changes below

```python
...
# from OCCT.WNT import WNT_Window
from OCCT.XwWindow import Xw_Window
...
from OCCT.Exchange import ExchangeBasic
...
try:
    from PySide2 import QtCore
    from PySide2.QtWidgets import QApplication, QMainWindow, QOpenGLWidget
    from PySide2.QtGui import QPalette, QIcon
...
class View(QOpenGLWidget):
...
        # Windows window
        # self.wind = WNT_Window(self.window_handle)
        self.wind = Xw_Window(self.display_connect, self.window_handle)
...
if __name__ == '__main__':
    shape = ExchangeBasic.read_step('./models/compressor.step')
    v = Viewer()
    v.display(shape,
              rgb=(0.5, 0.5, 0.5),
              material=Graphic3d_NOM_ALUMINIUM)
    v.start()
```
Here the full code

[Basic.py.zip](https://github.com/LaughlinResearch/pyOCCT/files/2342148/Basic.py.zip)

adam-urbanczyk, 2018-09-11:
I might have seen a similar issue with PythonOCC. Could you share a screenshot?

ffvpor, 2018-09-12:
During launching (terminal messages) and first screen (changed to 640w x 200h but same behavior as original sizes)
[image]
Note: I already seen GTK-Messages for other applications
After one click on window
[image]
or if I resize window, flashing view of compressor during moves. At the end of resizing, always black screen , fit view and now it's OK
[image]

jf---, 2018-09-12:
@ffvpor perhaps [this](https://github.com/tpaviot/pythonocc-core/blob/master/src/Display/wxDisplay.py#L53) is insightful. Perhaps your initializing the OCC window before Qt's OpenGL context is created or the Qt window is created?

trelau, 2018-09-13:
@ffvpor thanks for reporting. full disclaimer: the visualization tools were a bear minimum implementation to get to see *something*...there is definitely room for improvement and robustness. Hopefully someone on the thread can help. I'll try and get to it as I clear some other items off my list.

ffvpor, 2018-09-24:
@jf--- Thanks for clues, some improvements since your answers! I replaced QOpenGLWidget with QGLWidget (although to be avoided in new applications, starting from Qt 5.4) in the View class and changed the calling order of display connection, graphic driver, viewer and view functions. It is not perfect yet but there is better! 

Note: it's not the same step model

During launching, no TKOpenGL message

First screen: no black screen but very small model size

[image]

After a tiny resize, model fit all!

[image]

To be continued!
I will also work on a better understanding of using QOpenGLWidget

jf---, 2018-09-24:
@ffvpor that's wonderful news... perhaps time for a PR perhaps?
I ran into the exact same issue 

> "RuntimeError: Xw_Window, given invalid X window"

as you did...

___

For what its worth, I think pyside2 is a more robust choice over wxpython, its well maintained and supported and modern UI's can be build on Qt ( QML is pretty amazing ). Perhaps its something to consider moving fwd.

jf---, 2018-09-28:
I think that's a good idea. With PythonOCC we supported X, pyqt, pyside, and wxpython. I think settling on a clear choice ( and pyside is a very good one ) is a welcome idea. Also wxpython is more of a pain to support.

Perhaps @ffvpor can be tempted, such to be able advance things to a mergeable state?

ffvpor, 2018-09-30:
Many things to learn at the same time but I'am happy! With @jf--- advice, I managed to reach a functional code. First pull resquest done!

trelau, 2020-05-10:
@ffvpor master branch has been refreshed with occt 7.4.0 and some updated viewers (wx and qtpy that can use pyside2). would you be interested in testing it out? you can look at the actions here and the resulting conda packages https://github.com/LaughlinResearch/pyOCCT/actions/runs/97805754

maybe you can grab one and manually install and test out? either way, might be easiest to close this issue and reopen once i publish an "official" release? hopefully in next day or two...

closed 2020-05-12 by trelau