trelau/pyOCCT #19: compiling XwWindow on osx

jf---, 2018-09-12:
Hi,

I'm trying to compile `pyocct` on osx.
So far, so good, though compilation trips over `XwWindow`.

This is probably due to linking to `Xw_Window` rather than `Cocoa_Window`
[PythonOCC](https://github.com/tpaviot/pythonocc-core/blob/master/src/Visualization/Visualization.h#L23) does conditional compilation like so for the visualization module:

```c++
#ifdef WNT
  #include <windows.h>
  #include <WNT_WClass.hxx>
  #include <WNT_Window.hxx>
#elif defined(__APPLE__) && !defined(MACOSX_USE_GLX)
  #include <Cocoa_Window.hxx>
#else
  #include <Xw_Window.hxx>
#endif
```

Perhaps this could be carried over as:

```c++
#include <pyOCCT_Common.hpp>

#include <xTypes.h>

#include <Aspect_DisplayConnection.hxx>
#include <Aspect_FBConfig.hxx>
#include <Aspect_Window.hxx>

#ifdef WNT
  #include <WNT_Window.hxx>
#elif defined(__APPLE__) && !defined(MACOSX_USE_GLX)
  #include <Cocoa_Window.hxx>
#else
  #include <Xw_Window.hxx>
#endif

#include <Xw_Window.hxx>
```

Which brings me to my question; would by extension the class definition look like so?
( I'm new to `pybinder11` and not a `c++` programmer... go easy on me ;) 

```c++

   #ifdef WNT
     py::class_<Xw_Window, opencascade::handle<WNT_Window>, Aspect_Window>(mod, "Xw_Window", "This class defines XLib window intended for creation of OpenGL context.")
   #elif defined(__APPLE__) && !defined(MACOSX_USE_GLX)
     py::class_<Xw_Window, opencascade::handle<Cocoa_Window>, Aspect_Window>(mod, "Xw_Window", "This class defines XLib window intended for creation of OpenGL context.")
   #else
     py::class_<Xw_Window, opencascade::handle<Xw_Window>, Aspect_Window>(mod, "Xw_Window", "This class defines XLib window intended for creation of OpenGL context.")
   #endif
```

I'm a little surprised, since seeing `pyocct` up and running on windows, which should run into the same compilation issue I figure. 

Finally, [this thread](https://github.com/conda-forge/staged-recipes/pull/5937) on conda-forge was very helpful ( `MacOSX10.9.sdk` pointer ). That said, should those improvements to the conda recipe not be carried over to this repo?

jf---, 2018-09-13:
>  XwWindow: f15d989

That's a very helpful pointer, thx...

> ecause the build time was way beyond the limits for Appveyor and Travis (both were grateful enough to extend the build time to 3 hours I think for Laughlin Research, but I never asked for conda)

Thomas asked for a similar thing for `pythonocc`, on TravisCI.

Something really cool is that succesful compile's on master got uploaded to Thomas' conda account, which means that there's then very little overhead keeping up with changes in `master`.
I could look into porting that?

trelau, 2018-09-13:
CI integration hasn't been fully built out, so that would be a great next step. I think it makes sense to upload successful master builds to my account, and eventually I'd like to get conda-forge builds for releases.

trelau, 2020-05-10:
@jf--- master branch is updated to occt 7.4.0 (and smesh) and CI for windows and linux py 3.6 to 3.8. i experimented with osx support but didn't get the CI stuff right and hard to do since i don't have that platform locally to quickly debug.

if you're interested, i have a branch i could push with some github actions for osx if you want to give it a shot and/or provide some feedback. but even just building the bindings locally on osx would be a great help. actually...i'm going to push that osx branch but leave out the compile part, but it will generate the binding source. if you want, you can grab that and try and compile. any feedback or tweaks to the binding process would be great!