trelau/pyOCCT #21: install pyocct PackageNotFoundError

HuaJianDeng, 2018-12-11:
@trelau Hello,

I am a beginner to python. I found your pyOCCT very helpful so I want to use it. But some problems occured when I followed the instructions to install.  I work with win64.

conda create -n occt python=3.5
conda activate occt
conda install -c trelau pyocct

then, it showed:

1. Solving environment: failed

2.PackageNotFoundError: The following packages are not available from current channels:
  -pyocct
  -netgen ==6.2.1804
  -pyocct
  -occt ==7.2.0
  -pyocct
  -smesh == 8.3.0b0

Could you please offer me some suggestions ?  I will appreciate it very much.

_Originally posted by @HuaJianDeng in https://github.com/LaughlinResearch/pyOCCT/issue_comments#issuecomment-446118515_

trelau, 2018-12-11:
For win64 I recommend:

```conda install -c laughlinresearch -c conda-forge pyocct```

Please close the issue if this solves your problem.

HuaJianDeng, 2018-12-12:
Thank you ! @trelau  It has solved my problem.

closed 2018-12-12 by HuaJianDeng