trelau/pyOCCT #22: support for OCCT 7.3.0

Trophime, 2019-07-04:
Hi,
I'm wondering if you have any plans to support OCCT 7.3.0 and SMESH 9.1.9?

Best
C

trelau, 2019-07-04:
Hi,

The short answer is "yes to both". Although, these efforts were closely related to my full-time job but are now hobbies so it's been tough to find time. I've been trying to pick away at updating the binding generators for 7.3.0. If you're interested in helping out I could perhaps pull what I have together and maybe layout a roadmap for both projects if you'd like to contribute.

Trophime, 2019-07-05:
Hi,
after some struggle, I manage to get pyOCCT compile with OCCT 7.3.0.
What is the next step to test it?

Best

trelau, 2019-07-05:
Oh wow...how did you do that? just link to 7.3.0 and manually patch things up? i figured with any OCCT API changes I would need to re-run the binding generator...

For testing, I'd suggest at a minimum running the small set of unit tests in this repo. Another way might be to run the tests in this repo https://github.com/LaughlinResearch/AFEM. pyOCCT was initially built to support this project and wraps a lot of the functionality, so running the tests in that repo would be another good start.

kkremitzki, 2019-07-19:
@Trophime If you were to create a fork and push your changes there I would like to review and test as well.

trelau, 2019-07-19:
it's very much WIP...but pushed a branch of the binding generator i've been trying to pick away at...the goal is to capture all the edge cases in the binder...see the NOTES.txt for a rough plan...i think it compiles on Windows but doesn't actually fully work because i haven't put in some call guards and such for circular imports https://github.com/LaughlinResearch/pyOCCT_binder/blob/refactoring/NOTES.txt

trelau, 2019-08-19:
There is a branch for OCCT 7.3.0 available for testing

closed 2019-08-19 by trelau