trelau/pyOCCT #23: Build issues for 7.3.0 branch on Ubuntu

kkremitzki, 2019-08-20:
I am testing building the new 7.3.0 branch. It looks like there's a problem with something Windows-only:

```
[  6%] Building CXX object CMakeFiles/TCollection.dir/src/modules/TCollection/bind_TCollection_ExtendedString.cxx.o
/home/kkremitzki/.local/src/pyocct/src/modules/TCollection/bind_TCollection_ExtendedString.cxx: In function ‘void bind_TCollection_ExtendedString(pybind11::module&)’:
/home/kkremitzki/.local/src/pyocct/src/modules/TCollection/bind_TCollection_ExtendedString.cxx:104:149: error: ‘ToWideString’ is not a member of ‘TCollection_ExtendedString’
 cls_TCollection_ExtendedString.def("ToWideString", (const Standard_WideChar * (TCollection_ExtendedString::*)() const) &TCollection_ExtendedString::ToWideString, "Returns pointer to string as wchar_t* on Windows platform where wchar_t* is considered as UTF-16 string. This method is useful to pass string into wide-char system APIs, and makes sense only on Windows (other systems use UTF-8 and can miss wide-char functions at all).");
                                                                                                                                                     ^~~~~~~~~~~~
make[2]: *** [CMakeFiles/TCollection.dir/build.make:167: CMakeFiles/TCollection.dir/src/modules/TCollection/bind_TCollection_ExtendedString.cxx.o] Error 1
```

If I comment that out, I get:
```
[ 14%] Building CXX object CMakeFiles/ShapeAnalysis.dir/src/modules/ShapeAnalysis/bind_ShapeAnalysis_BoxBndTree.cxx.o
In file included from /home/kkremitzki/.local/src/pyocct/src/modules/ShapeAnalysis/bind_ShapeAnalysis_BoxBndTree.cxx:28:
/home/kkremitzki/.local/src/pyocct/src/include/bind_NCollection_UBTree.hxx: In function ‘void bind_NCollection_UBTree(pybind11::module&, const string&, const pybind11::module_local&)’:
/home/kkremitzki/.local/src/pyocct/src/include/bind_NCollection_UBTree.hxx:62:188: error: type/value mismatch at argument 1 in template parameter list for ‘template<class T> struct Deleter’
 py::class_<NCollection_UBTree<TheObjType, TheBndType>::Selector, std::unique_ptr<NCollection_UBTree<TheObjType, TheBndType>::Selector, Deleter<NCollection_UBTree<TheObjType, TheBndType>::Selector>>> cls_Selector(cls_NCollection_UBTree, "Selector", "Class defining the minimal interface of selector.", local);
                                                                                                                                                                                            ^~~~~~~~
/home/kkremitzki/.local/src/pyocct/src/include/bind_NCollection_UBTree.hxx:62:188: note:   expected a type, got ‘NCollection_UBTree<TheObjType, TheBndType>::Selector’
/home/kkremitzki/.local/src/pyocct/src/include/bind_NCollection_UBTree.hxx:62:196: error: type/value mismatch at argument 1 in template parameter list for ‘template<class _Tp, class _Dp> class std::unique_ptr’
 py::class_<NCollection_UBTree<TheObjType, TheBndType>::Selector, std::unique_ptr<NCollection_UBTree<TheObjType, TheBndType>::Selector, Deleter<NCollection_UBTree<TheObjType, TheBndType>::Selector>>> cls_Selector(cls_NCollection_UBTree, "Selector", "Class defining the minimal interface of selector.", local);
                                                                                                                                                                                                    ^~
/home/kkremitzki/.local/src/pyocct/src/include/bind_NCollection_UBTree.hxx:62:196: note:   expected a type, got ‘NCollection_UBTree<TheObjType, TheBndType>::Selector’
/home/kkremitzki/.local/src/pyocct/src/include/bind_NCollection_UBTree.hxx:62:196: error: template argument 2 is invalid
/home/kkremitzki/.local/src/pyocct/src/include/bind_NCollection_UBTree.hxx:62:198: error: type/value mismatch at argument 1 in template parameter list for ‘template<class type_, class ... options> class pybind11::class_’
 py::class_<NCollection_UBTree<TheObjType, TheBndType>::Selector, std::unique_ptr<NCollection_UBTree<TheObjType, TheBndType>::Selector, Deleter<NCollection_UBTree<TheObjType, TheBndType>::Selector>>> cls_Selector(cls_NCollection_UBTree, "Selector", "Class defining the minimal interface of selector.", local);
                                                                                                                                                                                                      ^
/home/kkremitzki/.local/src/pyocct/src/include/bind_NCollection_UBTree.hxx:62:198: note:   expected a type, got ‘NCollection_UBTree<TheObjType, TheBndType>::Selector’
/home/kkremitzki/.local/src/pyocct/src/include/bind_NCollection_UBTree.hxx:62:198: error: template argument 2 is invalid
/home/kkremitzki/.local/src/pyocct/src/include/bind_NCollection_UBTree.hxx:62:307: error: expression list treated as compound expression in initializer [-fpermissive]
 BTree<TheObjType, TheBndType>::Selector, Deleter<NCollection_UBTree<TheObjType, TheBndType>::Selector>>> cls_Selector(cls_NCollection_UBTree, "Selector", "Class defining the minimal interface of selector.", local);
                                                                                                                                                                                                                     ^

/home/kkremitzki/.local/src/pyocct/src/include/bind_NCollection_UBTree.hxx:67:14: error: request for member ‘def’ in ‘cls_Selector’, which is of non-class type ‘int’
 cls_Selector.def("Reject", (Standard_Boolean (NCollection_UBTree<TheObjType, TheBndType>::Selector::*)(const TheBndType &) const) &NCollection_UBTree<TheObjType, TheBndType>::Selector::Reject, "Rejection base on the bounding type.", py::arg(""));
              ^~~
/home/kkremitzki/.local/src/pyocct/src/include/bind_NCollection_UBTree.hxx:68:14: error: request for member ‘def’ in ‘cls_Selector’, which is of non-class type ‘int’
 cls_Selector.def("Accept", (Standard_Boolean (NCollection_UBTree<TheObjType, TheBndType>::Selector::*)(const TheObjType &)) &NCollection_UBTree<TheObjType, TheBndType>::Selector::Accept, "Confirm the object while making necessary tests on it. This method is called when the bounding box of the object conforms to the conditions (see Reject()). It is also supposed to keep record of accepted objects.", py::arg(""));
              ^~~
/home/kkremitzki/.local/src/pyocct/src/include/bind_NCollection_UBTree.hxx:69:14: error: request for member ‘def’ in ‘cls_Selector’, which is of non-class type ‘int’
 cls_Selector.def("Stop", (Standard_Boolean (NCollection_UBTree<TheObjType, TheBndType>::Selector::*)() const) &NCollection_UBTree<TheObjType, TheBndType>::Selector::Stop, "This condition is checked after each call to Accept().");
              ^~~
/home/kkremitzki/.local/src/pyocct/src/include/bind_NCollection_UBTree.hxx: In instantiation of ‘void bind_NCollection_UBTree(pybind11::module&, const string&, const pybind11::module_local&) [with TheObjType = int; TheBndType = Bnd_Box; std::__cxx11::string = std::__cxx11::basic_string<char>]’:
/home/kkremitzki/.local/src/pyocct/src/modules/ShapeAnalysis/bind_ShapeAnalysis_BoxBndTree.cxx:32:90:   required from here
/home/kkremitzki/.local/src/pyocct/src/include/bind_NCollection_UBTree.hxx:62:302: warning: left operand of comma operator has no effect [-Wunused-value]
 BTree<TheObjType, TheBndType>::Selector, Deleter<NCollection_UBTree<TheObjType, TheBndType>::Selector>>> cls_Selector(cls_NCollection_UBTree, "Selector", "Class defining the minimal interface of selector.", local);
                                                                                                                                                                                                                ^~~~~

/home/kkremitzki/.local/src/pyocct/src/include/bind_NCollection_UBTree.hxx:62:302: warning: right operand of comma operator has no effect [-Wunused-value]
/home/kkremitzki/.local/src/pyocct/src/include/bind_NCollection_UBTree.hxx:62:302: warning: right operand of comma operator has no effect [-Wunused-value]
/home/kkremitzki/.local/src/pyocct/src/include/bind_NCollection_UBTree.hxx:62:200: error: cannot convert ‘const pybind11::module_local’ to ‘int’ in initialization
 py::class_<NCollection_UBTree<TheObjType, TheBndType>::Selector, std::unique_ptr<NCollection_UBTree<TheObjType, TheBndType>::Selector, Deleter<NCollection_UBTree<TheObjType, TheBndType>::Selector>>> cls_Selector(cls_NCollection_UBTree, "Selector", "Class defining the minimal interface of selector.", l [... cut ...] al);
                                                                                                                                                                                                        ^~~~~~~~~~~~
make[2]: *** [CMakeFiles/ShapeAnalysis.dir/build.make:89: CMakeFiles/ShapeAnalysis.dir/src/modules/ShapeAnalysis/bind_ShapeAnalysis_BoxBndTree.cxx.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2034: CMakeFiles/ShapeAnalysis.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

trelau, 2019-08-22:
I updated the bindings in latest push to remove the "Deleter" template entirely as it's not needed anymore due to pybind11 update. Maybe that will help?

I need to add some logic in the binder to comment out the windows specific stuff, but maybe the latest commit will resolve the other issues?

trelau, 2019-08-24:
Excluded TCollection_ExtendedString::ToWideString in latest occt_730 commit

kkremitzki, 2019-09-03:
Current progress with the latest changes:
```
[ 19%] Built target ShapeExtend
[ 20%] Building CXX object CMakeFiles/ShapeAnalysis.dir/src/modules/ShapeAnalysis.cxx.o
In file included from /home/kkremitzki/.local/src/pyocct/github/src/modules/ShapeAnalysis.cxx:102:
/home/kkremitzki/.local/src/pyocct/github/src/templates/bind_NCollection_UBTree.hxx: In function ‘void bind_NCollection_UBTree(pybind11::module&, const string&, const pybind11::module_local&)’:
/home/kkremitzki/.local/src/pyocct/github/src/templates/bind_NCollection_UBTree.hxx:57:64: error: type/value mismatch at argument 1 in template parameter list for ‘template<class type_, class ... options> class pybind11::class_’
 py::class_<NCollection_UBTree<TheObjType, TheBndType>::Selector> cls_NCollection_UBTree_Selector(cls_NCollection_UBTree, "Selector", "Class defining the minimal interface of selector.", local);
                                                                ^
/home/kkremitzki/.local/src/pyocct/github/src/templates/bind_NCollection_UBTree.hxx:57:64: note:   expected a type, got ‘NCollection_UBTree<TheObjType, TheBndType>::Selector’
/home/kkremitzki/.local/src/pyocct/github/src/templates/bind_NCollection_UBTree.hxx:57:192: error: expression list treated as compound expression in initializer [-fpermissive]
 py::class_<NCollection_UBTree<TheObjType, TheBndType>::Selector> cls_NCollection_UBTree_Selector(cls_NCollection_UBTree, "Selector", "Class defining the minimal interface of selector.", local);
                                                                                                                                                                                                ^
/home/kkremitzki/.local/src/pyocct/github/src/templates/bind_NCollection_UBTree.hxx:60:33: error: request for member ‘def’ in ‘cls_NCollection_UBTree_Selector’, which is of non-class type ‘int’
 cls_NCollection_UBTree_Selector.def("Reject", (Standard_Boolean (NCollection_UBTree<TheObjType, TheBndType>::Selector::*)(const TheBndType &) const) &NCollection_UBTree<TheObjType, TheBndType>::Selector::Reject, "Rejection base on the bounding type.", py::arg(""));
                                 ^~~
/home/kkremitzki/.local/src/pyocct/github/src/templates/bind_NCollection_UBTree.hxx:61:33: error: request for member ‘def’ in ‘cls_NCollection_UBTree_Selector’, which is of non-class type ‘int’
 cls_NCollection_UBTree_Selector.def("Accept", (Standard_Boolean (NCollection_UBTree<TheObjType, TheBndType>::Selector::*)(const TheObjType &)) &NCollection_UBTree<TheObjType, TheBndType>::Selector::Accept, "Confirm the object while making necessary tests on it. This method is called when the bounding box of the object conforms to the conditions (see Reject()). It is also supposed to keep record of accepted objects.", py::arg(""));
                                 ^~~
/home/kkremitzki/.local/src/pyocct/github/src/templates/bind_NCollection_UBTree.hxx:62:33: error: request for member ‘def’ in ‘cls_NCollection_UBTree_Selector’, which is of non-class type ‘int’
 cls_NCollection_UBTree_Selector.def("Stop", (Standard_Boolean (NCollection_UBTree<TheObjType, TheBndType>::Selector::*)() const) &NCollection_UBTree<TheObjType, TheBndType>::Selector::Stop, "This condition is checked after each call to Accept().");
                                 ^~~
/home/kkremitzki/.local/src/pyocct/github/src/templates/bind_NCollection_UBTree.hxx: In instantiation of ‘void bind_NCollection_UBTree(pybind11::module&, const string&, const pybind11::module_local&) [with TheObjType = int; TheBndType = Bnd_Box; std::__cxx11::string = std::__cxx11::basic_string<char>]’:
/home/kkremitzki/.local/src/pyocct/github/src/modules/ShapeAnalysis.cxx:146:90:   required from here
/home/kkremitzki/.local/src/pyocct/github/src/templates/bind_NCollection_UBTree.hxx:57:187: warning: left operand of comma operator has no effect [-Wunused-value]
 py::class_<NCollection_UBTree<TheObjType, TheBndType>::Selector> cls_NCollection_UBTree_Selector(cls_NCollection_UBTree, "Selector", "Class defining the minimal interface of selector.", local);
                                                                                                                                                                                           ^~~~~
/home/kkremitzki/.local/src/pyocct/github/src/templates/bind_NCollection_UBTree.hxx:57:187: warning: right operand of comma operator has no effect [-Wunused-value]
/home/kkremitzki/.local/src/pyocct/github/src/templates/bind_NCollection_UBTree.hxx:57:187: warning: right operand of comma operator has no effect [-Wunused-value]
/home/kkremitzki/.local/src/pyocct/github/src/templates/bind_NCollection_UBTree.hxx:57:66: error: cannot convert ‘const pybind11::module_local’ to ‘int’ in initialization
 py::class_<NCollection_UBTree<TheObjType, TheBndType>::Selector> cls_NCollection_UBTree_Selector(cls_NCollection_UBTree, "Selector", "Class defining the minimal interface of selector.", local);
                                                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/ShapeAnalysis.dir/build.make:63: CMakeFiles/ShapeAnalysis.dir/src/modules/ShapeAnalysis.cxx.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2071: CMakeFiles/ShapeAnalysis.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

trelau, 2019-09-06:
hmm...seems like it's related to sub-class of a templated type...wonder if there needs to be a `typename` qualifier somewhere? something related to this commit? https://github.com/LaughlinResearch/pyOCCT/commit/ba97e5e7917b7814a504a4fdab837bba636077b4

trelau, 2020-05-03:
OCCT 7.4.0 and better support for ubuntu part of https://github.com/LaughlinResearch/pyOCCT/pull/41

closed 2020-05-03 by trelau