trelau/pyOCCT #24: Build and Install Documentation

jimshepherd, 2019-09-05:
First, thank you for developing and providing this package!

There are some bugs fixed in the 7.3 release of OCC that address some issues we have run into. I would like to try your occt_730 branch, but I'm not sure how to build and install the software for the Conda and Win64 environment. I read one of the earlier issues in which building was discussed but could not figure out what needs to be done. Do you have documentation on the building process and making the built package available in Anaconda? Thanks for any help you can provide.

trelau, 2019-09-06:
the occt_730 bracnh is a work in progress so build and install docs are on the to do list...but...i did get a successful appveyor build the other night so maybe you can download the artifact and install locally if you're just looking to test things out? https://ci.appveyor.com/project/LaughlinResearch/pyocct/builds/27111608/artifacts

if you want to build yourself, you might clone the repo and use `conda build ci/conda`. if successful, it should build a similar artifact...i think this kind of describes how to install locally https://stackoverflow.com/questions/34316951/install-anaconda-library-from-a-local-source

otherwise, you can do the traditional `cmake -G"Visual Studio 14 Win64" ..` from say a `build` folder you create after cloning the repo...i do this locally use a conda environment called `pyocct` where i've installed all the dependencies. usually i'll run that first, then launch `cmake-gui ..` to get the GUI and generate it from there. For now, it should mostly just work but you need to define `TBB_INCLUDE_DIR` manually for now...i can't seem to get `find_package(TBB)` to work yet...

jimshepherd, 2019-09-06:
Thank you! The artifact you provided worked well. Just had to upgrade the occt package to 7.3.0 and the 730 branch works. There are some methods not available in this branch which I will open another issue for.

trelau, 2020-05-10:
the README has been updated with install and build from sources. check it out!

closed 2020-05-10 by trelau