trelau/pyOCCT #25: BOPAlgo_Options protected methods not available

jimshepherd, 2019-09-06:
In the occt_730 branch, protected methods from the BOPAlgo_Options class, e.g. SetFuzzyValue, HasWarnings, are not available to instances of classes that inherit it, e.g. BRepAlgoAPI_Fuse.

In the master branch, these protected methods are defined at the top of the BRepAlgoAPI.cpp file with the following comment:
// FIXME Access protected members of this class that are made available with "using". Need to write as lambdas.

Should I just add a class definition for BOPAlgo_Options similarly to the other classes defined in the BRepAlgoAPI.cxx file or does the protected nature of these methods need to be handled differently?

trelau, 2019-09-06:
it will likely need a similar update as shown here in master https://github.com/LaughlinResearch/pyOCCT/commit/4d49f20ec8672924f6f484c346e4cf9c5a51e591

i just haven't gotten that far in the binding generator yet...it's on he to-do list

trelau, 2020-05-10:
this is now covered by unit tests and captured in binding generator for the future https://github.com/LaughlinResearch/pyOCCT/blob/4becc643c4ee0c3babbc01a9ffd1edafbe86b68d/test/test_BRepAlgoAPI.py#L57

closed 2020-05-10 by trelau