trelau/pyOCCT #30: CQ + LR = love ? ...

Andrei-Pozolotin, 2020-04-03:
I am curious if there is enough synergy / interest to join forces / reduce duplication:

CQ:
https://github.com/CadQuery/pywrap
https://github.com/CadQuery/OCP

LR:
https://github.com/LaughlinResearch/pyOCCT
https://github.com/LaughlinResearch/pyOCCT_binder

trelau, 2020-05-03:
it's a great project. for now, i'm focusing on pyOCCT as a fun hobby, so staying focused on stabilizing it for multiple platforms and continuing to provide support for SMESH

closed 2020-05-03 by trelau

adam-urbanczyk, 2020-05-08:
Good to hear you are back @trelau . Let's see in the future for some synergy. I'm looking into wrapping the VTK support modules of OCCT, maybe it'll be interesting for you too.