trelau/pyOCCT #31: conda install incompatible Python version

bakakouhai, 2020-04-23:
Hello. I am trying to install pyocct on Pop_OS 18.04. I have installed anaconda and create conda environment with python 3.5, 3.6, and 3.7. All of them failed saying incompatible Python version.

The command prompt was written below. May I know which specific version should work for current open cascade version?

Kind regards, Iwan

Upon installing pyocct, the process failed as follows:
```
(occt) iwan@pop-os:~$ conda install -c trelau pyocct
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: \ 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                          

UnsatisfiableError: The following specifications were found
to be incompatible with the existing python installation in your environment:

Specifications:

  - pyocct -> python[version='>=2.7,<2.8.0a0|>=3.7,<3.8.0a0']

Your python: python=3.5

If python is on the left-most side of the chain, that's the version you've asked for.
When python appears to the right, that indicates that the thing on the left is somehow
not available for the python version you are constrained to. Note that conda will not
change your python version to a different minor version unless you explicitly specify
that.
```

It seems the python required is 3.7, so I create another environment with python3.7 and do the same process. It end up like this:
```
(occt_3.7) iwan@pop-os:~$ conda install -c trelau pyocct
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: - 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                          

UnsatisfiableError: The following specifications were found
to be incompatible with the existing python installation in your environment:

Specifications:

  - pyocct -> python[version='>=3.5,<3.6.0a0|>=3.6,<3.7.0a0']

Your python: python=3.7

If python is on the left-most side of the chain, that's the version you've asked for.
When python appears to the right, that indicates that the thing on the left is somehow
not available for the python version you are constrained to. Note that conda will not
change your python version to a different minor version unless you explicitly specify
that.
```

So this time it asked for version 3.6., which ends up asking for 3.7 version:
```
(occt_3.6) iwan@pop-os:~$ conda install -c trelau pyocct
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: \ 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                          

UnsatisfiableError: The following specifications were found
to be incompatible with the existing python installation in your environment:

Specifications:

  - pyocct -> python[version='>=2.7,<2.8.0a0|>=3.7,<3.8.0a0']

Your python: python=3.6

If python is on the left-most side of the chain, that's the version you've asked for.
When python appears to the right, that indicates that the thing on the left is somehow
not available for the python version you are constrained to. Note that conda will not
change your python version to a different minor version unless you explicitly specify
that.
```

frmdstryr, 2020-04-24:
The version of `pyOCCT` on conda only is built for python 3.6. I remember getting this before and If I remember correctly the "fix" was to install both at the same time. Eg

```
conda install -c trelau pyocct python=3.6
```

bakakouhai, 2020-04-24:
@frmdstryr Thank you for the fast reply.
Following your suggestion, I deleted all conda environments to start from scratch.
New environment was created afterwards without any python environement, i.e. :
`iwan@pop-os:~$ conda create -n occt_3.6`

I then run the installation code as suggested. It still get the same error.
```
(occt_3.6) iwan@pop-os:~$ conda install -c trelau pyocct python=3.6
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: \ 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                          

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versions

Package python conflicts for:
python=3.6
pyocct -> python[version='>=3.5,<3.6.0a0|>=3.6,<3.7.0a0']
pyocct -> vtk==8.1.0 -> python[version='>=2.7,<2.8.0a0|>=3.7,<3.8.0a0']
```

frmdstryr, 2020-04-24:
Is conda-forge added as a channel in your config? If not also pass `-c conda-forge`

bakakouhai, 2020-04-24:
Thank you. That is the problem. I can install it now.

closed 2020-04-24 by bakakouhai