trelau/pyOCCT #32: %run 'import_step.py' result in libatk and wx error

bakakouhai, 2020-04-24:
Hello.
I have successfully installed pyocct now. 
Python 3.5 was used in OCCT env following the installation page.
nb_conda_kernel was installed in conda base env to access other env.

`from OCCT.TopoDS import TopoDS_Shape` returns no error

However,
`%run 'import_step.py'` gives the following error
```
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
~/Documents/pyOCCT-master/examples/import_step.py in <module>
     20 
     21 from OCCT.Exchange import ExchangeBasic
---> 22 from OCCT.Visualization import BasicViewer
     23 
     24 shape = ExchangeBasic.read_step('./models/compressor.step')

~/anaconda3/envs/occt/lib/python3.5/site-packages/OCCT/Visualization/__init__.py in <module>
     17 # License along with this library; if not, write to the Free Software
     18 # Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston, MA  02110-1301 USA
---> 19 from OCCT.Visualization.Basic import *

~/anaconda3/envs/occt/lib/python3.5/site-packages/OCCT/Visualization/Basic.py in <module>
     22 import platform
     23 
---> 24 import wx
     25 from OCCT.AIS import (AIS_InteractiveContext, AIS_Shape, AIS_Shaded,
     26                       AIS_WireFrame)

~/anaconda3/envs/occt/lib/python3.5/site-packages/wx/__init__.py in <module>
     15 # Import all items from the core wxPython module so they appear in the wx
     16 # package namespace.
---> 17 from wx.core import *
     18 
     19 

~/anaconda3/envs/occt/lib/python3.5/site-packages/wx/core.py in <module>
     10 """
     11 
---> 12 from ._core import *
     13 
     14 #-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=

ImportError: /usr/lib/x86_64-linux-gnu/libatk-1.0.so.0: undefined symbol: g_log_structured_standard
```

running the script for the second time (without resetting the kernel) gives the following error:
```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
~/Documents/pyOCCT-master/examples/import_step.py in <module>
     20 
     21 from OCCT.Exchange import ExchangeBasic
---> 22 from OCCT.Visualization import BasicViewer
     23 
     24 shape = ExchangeBasic.read_step('./models/compressor.step')

~/anaconda3/envs/occt/lib/python3.5/site-packages/OCCT/Visualization/__init__.py in <module>
     17 # License along with this library; if not, write to the Free Software
     18 # Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston, MA  02110-1301 USA
---> 19 from OCCT.Visualization.Basic import *

~/anaconda3/envs/occt/lib/python3.5/site-packages/OCCT/Visualization/Basic.py in <module>
     22 import platform
     23 
---> 24 import wx
     25 from OCCT.AIS import (AIS_InteractiveContext, AIS_Shape, AIS_Shaded,
     26                       AIS_WireFrame)

~/anaconda3/envs/occt/lib/python3.5/site-packages/wx/__init__.py in <module>
     10 # Load the main version string into the package namespace
     11 import wx.__version__
---> 12 __version__ = wx.__version__.VERSION_STRING
     13 
     14 

AttributeError: module 'wx' has no attribute '__version__'
```

I think this error is related to visualisation issue?
Is there a specific version for wxpython that I need to install?
Synaptic shows libatk is installed.

Kind regards,
Iwan

trelau, 2020-04-25:
have you installed wx by something like `conda install -c conda-forge wxpython`?

bakakouhai, 2020-04-25:
Thanks for the reply.

I did install the wxpython inside the occt env.
```
(occt) iwan@pop-os:~$ pip list
Package          Version  
---------------- ---------
backcall         0.1.0    
certifi          2018.8.24
decorator        4.4.2    
future           0.16.0   
ipykernel        5.1.0    
ipython          7.0.1    
ipython-genutils 0.2.0    
jedi             0.12.1   
jupyter-client   5.3.3    
jupyter-core     4.5.0    
OCCT             7.2.0.1  
parso            0.7.0    
pexpect          4.6.0    
pickleshare      0.7.5    
pip              20.0.2   
prompt-toolkit   2.0.10   
ptyprocess       0.6.0    
Pygments         2.6.1    
python-dateutil  2.8.1    
pyzmq            17.1.2   
setuptools       40.4.3   
simplegeneric    0.8.1    
six              1.14.0   
tornado          5.1.1    
traitlets        4.3.2    
wcwidth          0.1.9    
wheel            0.34.2   
wxPython         4.0.1
```

trelau, 2020-05-10:
can you try this on the latest builds? download the relevant conda package and install it manually to test and let me know https://github.com/LaughlinResearch/pyOCCT/actions/runs/97805754

bakakouhai, 2020-05-11:
Hello. I have downloaded pyocct for ubuntu 3.7. How do I install it to conda environment?
I have tried using `conda install --offline` but it does not work. Also, does the extension needs to be .tar.bz2 ?

frmdstryr, 2020-05-11:
CI bundles the whole build folder so you have to unzip it (or just extract the pyocct package from the `linux-64` folder), then just run:

```sh
conda install path/to/pyocct-7.4.0.0-py37_0.tar.bz2
```

bakakouhai, 2020-05-11:
@frmdstryr Thank you for the instruction. I point the installer to the extracted folder before so conda keep looking up online for the package.

@trelau I have installed pyocct, wxpython and jupyterlab on pyocct env.
I split the import_step.py into three cells according to the description:
1. import libraries --> run successfully
2. # Read the file and get the shape --> run successfully
3. # Explore the faces of the shape (these are known to be named) --> error

The output of the third cell is as follows:
```
Found entity: CYLINDER_TOP
Found entity: FACE_UP
Press "c" to continue...

---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
~/anaconda3/envs/pyocct/lib/python3.7/site-packages/wx/core.py in <lambda>(event)
   3283         app._CallAfterId = wx.NewEventType()
   3284         app.Connect(-1, -1, app._CallAfterId,
-> 3285                     lambda event: event.callable(*event.args, **event.kw) )
   3286     evt = wx.PyEvent()
   3287     evt.SetEventType(app._CallAfterId)

~/anaconda3/envs/pyocct/lib/python3.7/site-packages/OCCT/Visualization/WxViewer.py in _init_wind(self, fit, background)
    138                 wind = WNT_Window(capsule)
    139             elif sys.platform.startswith('linux'):
--> 140                 from OCCT.XwWindow import Xw_Window
    141 
    142                 # Xw window

ModuleNotFoundError: No module named 'OCCT.XwWindow'
```

frmdstryr, 2020-05-11:
Looks like that was not updated... it should be `from OCCT.Xw import Xw_Window`

closed 2020-05-11 by trelau

bakakouhai, 2020-05-24:
@frmdstryr Hello. Sorry for the late response. 
I have downloaded the dev build from [here](https://github.com/LaughlinResearch/pyOCCT/actions/runs/101800932) and install it on a new environment.

I get Xw window error as follows:
```
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
~/anaconda3/envs/pyocct/lib/python3.7/site-packages/wx/core.py in <lambda>(event)
   3283         app._CallAfterId = wx.NewEventType()
   3284         app.Connect(-1, -1, app._CallAfterId,
-> 3285                     lambda event: event.callable(*event.args, **event.kw) )
   3286     evt = wx.PyEvent()
   3287     evt.SetEventType(app._CallAfterId)

~/anaconda3/envs/pyocct/lib/python3.7/site-packages/OCCT/Visualization/WxViewer.py in _init_wind(self, fit, background)
    141 
    142                 # Xw window
--> 143                 wind = Xw_Window(self.display_connect, hwnd)
    144             else:
    145                 raise NotImplementedError('Support platform not found for visualization.')

RuntimeError: Xw_Window, given invalid X window
```

trelau, 2020-05-24:
@bakakouhai just to be sure, can you try one of these packages that i uploaded recently for testing? https://anaconda.org/trelau/pyocct

bakakouhai, 2020-05-24:
@trelau still getting the same error. Conda list is listed below.
```
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
~/anaconda3/envs/pyocct/lib/python3.7/site-packages/wx/core.py in <lambda>(event)
   3283         app._CallAfterId = wx.NewEventType()
   3284         app.Connect(-1, -1, app._CallAfterId,
-> 3285                     lambda event: event.callable(*event.args, **event.kw) )
   3286     evt = wx.PyEvent()
   3287     evt.SetEventType(app._CallAfterId)

~/anaconda3/envs/pyocct/lib/python3.7/site-packages/OCCT/Visualization/WxViewer.py in _init_wind(self, fit, background)
    141 
    142                 # Xw window
--> 143                 wind = Xw_Window(self.display_connect, hwnd)
    144             else:
    145                 raise NotImplementedError('Support platform not found for visualization.')

RuntimeError: Xw_Window, given invalid X window
```

```
iwan@pop-os:~$ source activate
(base) iwan@pop-os:~$ conda activate pyocct
(pyocct) iwan@pop-os:~$ conda list
# packages in environment at /home/iwan/anaconda3/envs/pyocct:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       0_gnu    conda-forge
atk                       2.36.0                        0    conda-forge
atk-1.0                   2.36.0               haf93ef1_0    conda-forge
attrs                     19.3.0                     py_0    conda-forge
backcall                  0.1.0                      py_0    conda-forge
binutils_impl_linux-64    2.34                 h53a641e_2    conda-forge
bleach                    3.1.5              pyh9f0ad1d_0    conda-forge
boost-cpp                 1.72.0               h8e57a91_0    conda-forge
brotlipy                  0.7.0           py37h8f50634_1000    conda-forge
bzip2                     1.0.8                h516909a_2    conda-forge
ca-certificates           2020.4.5.1           hecc5488_0    conda-forge
cairo                     1.16.0            hcf35c78_1003    conda-forge
certifi                   2020.4.5.1       py37hc8dfbb8_0    conda-forge
cffi                      1.14.0           py37hd463f26_0    conda-forge
chardet                   3.0.4           py37hc8dfbb8_1006    conda-forge
cryptography              2.9.2            py37hb09aad4_0    conda-forge
curl                      7.69.1               h33f0ec9_0    conda-forge
decorator                 4.4.2                      py_0    conda-forge
defusedxml                0.6.0                      py_0    conda-forge
entrypoints               0.3             py37hc8dfbb8_1001    conda-forge
expat                     2.2.9                he1b5a44_2    conda-forge
fontconfig                2.13.1            h86ecdb6_1001    conda-forge
freetype                  2.10.2               he06d7ca_0    conda-forge
fribidi                   1.0.9                h516909a_0    conda-forge
future                    0.18.2           py37hc8dfbb8_1    conda-forge
gcc_impl_linux-64         7.5.0                hd420e75_6    conda-forge
gdk-pixbuf                2.38.2               h3f25603_3    conda-forge
gettext                   0.19.8.1          hc5be6a0_1002    conda-forge
gfortran_impl_linux-64    7.5.0                hdf63c60_6    conda-forge
glib                      2.64.3               h6f030ca_0    conda-forge
gobject-introspection     1.58.2          py37h619baee_1004    conda-forge
graphite2                 1.3.13            he1b5a44_1001    conda-forge
gst-plugins-base          1.14.5               h0935bb2_2    conda-forge
gstreamer                 1.14.5               h36ae1b5_2    conda-forge
gtk2                      2.24.32              h586f36d_1    conda-forge
gxx_impl_linux-64         7.5.0                hdf63c60_6    conda-forge
harfbuzz                  2.4.0                h9f30f68_3    conda-forge
hdf4                      4.2.13            hf30be14_1003    conda-forge
hdf5                      1.10.6          nompi_h3c11f04_100    conda-forge
icu                       64.2                 he1b5a44_1    conda-forge
idna                      2.9                        py_1    conda-forge
[... 86 lines cut ...]
urllib3                   1.25.9                     py_0    conda-forge
vtk                       8.2.0           py37h2bd422c_217    conda-forge
wcwidth                   0.1.9              pyh9f0ad1d_0    conda-forge
webencodings              0.5.1                      py_1    conda-forge
wheel                     0.34.2                     py_1    conda-forge
wxpython                  4.0.7.post2      py37h17634e0_3    conda-forge
xorg-kbproto              1.0.7             h14c3975_1002    conda-forge
xorg-libice               1.0.10               h516909a_0    conda-forge
xorg-libsm                1.2.3             h84519dc_1000    conda-forge
xorg-libx11               1.6.9                h516909a_0    conda-forge
xorg-libxau               1.0.9                h14c3975_0    conda-forge
xorg-libxdmcp             1.1.3                h516909a_0    conda-forge
xorg-libxext              1.3.4                h516909a_0    conda-forge
xorg-libxrender           0.9.10            h516909a_1002    conda-forge
xorg-libxt                1.2.0                h516909a_0    conda-forge
[... cut ...]
xorg-xextproto            7.3.0             h14c3975_1002    conda-forge
xorg-xproto               7.0.31            h14c3975_1007    conda-forge
xz                        5.2.5                h516909a_0    conda-forge
zeromq                    4.3.2                he1b5a44_2    conda-forge
zipp                      3.1.0                      py_0    conda-forge
zlib                      1.2.11            h516909a_1006    conda-forge
zstd                      1.4.4                h6597ccf_3    conda-forge

```

frmdstryr, 2020-05-24:
Does pop-os use wayland? 

With wayland you need to run `export QT_QPA_PLARFORM=xcb` for the `QtViewer`, wx may need something similar.

bakakouhai, 2020-05-24:
@frmdstryr I checked the custom.conf in my system based on [this](https://www.reddit.com/r/pop_os/comments/75wn18/how_do_i_start_wayland_session/) post. The WaylandEnable=False is not commented, so the system is running on X.

Is this an OS specific problem then?
[image]

frmdstryr, 2020-05-25:
Not sure. Have you tried the QtViewer? I know that works on ubuntu.

bakakouhai, 2020-05-25:
@frmdstryr Which package should I install to try the QtViewer?
From synaptic or conda install?
Sorry for the noob questions.

bakakouhai, 2020-05-26:
@frmdstryr I tried to install qt creator via synaptic and rerun the script. 
Nothing is changed. However, I see an error message below in my jupyter terminal:
```
Gtk-Message: 00:39:40.905: Failed to load module "atk-bridge"
Gtk-Message: 00:39:40.905: Failed to load module "appmenu-gtk-module"
Gtk-Message: 00:39:40.907: Failed to load module "canberra-gtk-module"
```
Did you get the error when running in ubuntu?
The error happened when running the part below:
```
# Read the file and get the shape
reader = STEPControl_Reader()
tr = reader.WS().TransferReader()
reader.ReadFile('./models/shape_names.step')
reader.TransferRoots()
shape = reader.OneShape()

gui = ViewerWx()
```

bakakouhai, 2020-06-14:
@frmdstryr Can you tell me which ubuntu do you have it working? I have tried installing pyocct on fresh 20.04 on Parallel Desktop. The same error still occurred.

Kind regards,
Iwan

frmdstryr, 2020-06-14:
18.04, I only tried the QtViewer

bakakouhai, 2020-06-14:
> 18.04, I only tried the QtViewer

Will try that this time then.

bakakouhai, 2020-06-15:
Tried the package in 18.04. the same error is there.