trelau/pyOCCT #40: Apply fix for Font_BRepFont

trelau, 2020-05-02:
from https://github.com/LaughlinResearch/pyOCCT/pull/38#issuecomment-622947680 @frmdstryr

> Awesome, looks like its working now!
> 
> Could you also pull in the changes to make Font_BRepFont work?
> 
> Add a subclass that exposes new and delete
> https://github.com/frmdstryr/pyOCCT/blob/github-actions/inc/bind_Font_BRepFont.hxx
> 
> Add header (and remove it from excluded classes)
> https://github.com/frmdstryr/pyOCCT/blob/github-actions/config.txt#L637
> 
> Patch Font_BRepTextBuilder to use the font subclass
> https://github.com/frmdstryr/pyOCCT/blob/github-actions/config.txt#L942-L946
> 
> I'm not sure if the nodelete option is needed there. I don't get why they decided to make it inherit from a protected class...

to do:
- apply recommended fixes
- write unit test to verify fix stays fixed as things update and/or change

frmdstryr, 2020-05-03:
Thanks for creating this, I probably should have just created an issue in the first place.  

Are you planning on merging into master soon or should I create a PR with this on top of the `occt_740` branch?

trelau, 2020-05-03:
getting a master PR set up right now

frmdstryr, 2020-05-03:
Should the binder config.txt be moved into this repo?

trelau, 2020-05-03:
yes, that is the plan. these are some my next steps:
- move config.txt to this repo and adjust anything in pyOCCT_binder to make it a bit more "neutral"
- Build bindings for each target platform only once. That is, build a conda environment with all the needed sources and header files, but run it only once for Windows, then use those sources for all python binding compiling (i.e., 3.6, 3.7, and 3.8). I don't mind regenerating the bindings each time as it is now, but it seems some of the dependencies are available for windows python 3.8, but you don't need 3.8 to simply generate the binding source code.
- support for osx
- potentially some automatic upload of master/dev packages and each release to laughlinresearch anaconda channel

closed 2020-05-03 by trelau