trelau/pyOCCT #42: Feature request: Support raytracing

frmdstryr, 2020-05-03:
I'm not sure if raytracing works on other platforms or not but it currently doesn't in linux (I tried with a Windows VM and it didn't work there either).  

I believe this is an issue with occt, but pyOCCT will also need to add support as well (if occt gets it).

## Expected Behavior

In the cad assistant app rendering with raytracing works (notice shadows, antialiasing, and reflections). 

[image]

## Current Behavior

In DeclaraCAD these settings have no effect as the OpenGl_GraphicDriver says it's not supported.

[image]

## Possible Solution

From scanning the code it looks like OCCT requires the "USE_GLES2"  flag then it imports the gles2 headers and enables this block: https://github.com/Open-Cascade-SAS/OCCT/blob/master/src/OpenGl/OpenGl_Context.cxx#L1707

I'm trying to get it to work on linux, I've managed to build the `occt` package with gles2 support but still need to get pyOCCT to work with it.

frmdstryr, 2020-05-06:
Heh, so this has nothing to do with the GLES2 flag...  

The problem was that pybind is making a copy of the returned object instead of the reference so the changes to the rendering params don't update the actual views parameters!

[image]

```c
# Return ref
+patch V3d: &V3d_View::ChangeRenderingParams-->&V3d_View::ChangeRenderingParams, py::return_value_policy::reference
```

Think binder should use `py::return_value_policy::reference` for anything that returns a non-const reference?

closed 2020-05-09 by trelau

frmdstryr, 2020-05-14:
I think detecting when a ref should be used needs to be added into the binder... a lot of methods return references (eg https://dev.opencascade.org/doc/refman/html/class_prs3d___drawer.html) that return copies and do nothing when modified from python (currently need to save a ref to the copy then reassign it). 

I'll try to to a PR.

trelau, 2020-05-14:
@frmdstryr had that in mind when making this ticket for ref https://github.com/LaughlinResearch/pyOCCT_binder/issues/14

PR #51 (2020-05-09): Add patch for V3d_View so ChangeRenderParams returns a reference
Fixes #42 

## Description

V3d_View::ChangeRenderParams returns a reference that needs modified to update the rendering params. By default pybind creates a copy, this patches the function to use a reference.

## Motivation and Context

The rending params currently can't be changed as it modifies a copy.

## How Has This Been Tested?

Built the module and tried it locally.

## Screenshots (if appropriate):

Added in #42 

## Types of changes

- [X] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking [... cut]