trelau/pyOCCT #54: ubuntu-latest py3.6 action doesn't succeed anymore

trelau, 2020-05-10:
why? https://github.com/LaughlinResearch/pyOCCT/runs/660835362?check_suite_focus=true

trelau, 2020-05-10:
@frmdstryr any ideas?

frmdstryr, 2020-05-10:
It's picking up 3.7 from linuxbrew for some reason... never heard of linuxbrew before. 

```
 -- Searching for pybind11...
-- Found PythonInterp: /home/linuxbrew/.linuxbrew/bin/python3.7 (found version "3.7.7") 
-- Found PythonLibs: /home/linuxbrew/.linuxbrew/opt/python/lib/libpython3.7m.so
```

Another +1 for building with docker to get an image without random junk in it.

trelau, 2020-05-10:
agreed on docker. was looking at that yesterday because that is how the occt feedstock is built for conda-forge...didn't quite find a good reference starting point. but might help linux and macos both?

frmdstryr, 2020-05-10:
Or just adding a `rm -Rf /home/linuxbrew/` step in ci would probably work.

trelau, 2020-05-11:
Fixed by #56

closed 2020-05-11 by trelau