trelau/pyOCCT #60: RuntimeError: Xw_Window, given invalid X window on Linux (Mint 19.2)

mnesarco, 2020-07-07:
Examples does not run in Linux:

```
(pyocct) dev@devel03:examples$ python bop_speed.py 
Gtk-Message: 17:08:52.114: Failed to load module "atk-bridge"
05:08:52 PM: Debug: Failed to connect to session manager: None of the authentication protocols specified are supported
Press "c" to continue...

(bop_speed.py:1986): Gdk-WARNING **: 17:08:52.451: gdkdrawable-x11.c:952 drawable is not a pixmap or window
Traceback (most recent call last):
  File "/home/mnesarco/miniconda3/envs/pyocct/lib/python3.8/site-packages/wx/core.py", line 3285, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "/home/mnesarco/miniconda3/envs/pyocct/lib/python3.8/site-packages/OCCT/Visualization/WxViewer.py", line 144, in _init_wind
    wind = Xw_Window(self.display_connect, hwnd)
RuntimeError: Xw_Window, given invalid X window
```

## Expected Behavior

Examples should run

## Steps to Reproduce (for bugs)

`python bop_speed.py`

## Your Environment

* Version used: 7.4.0
* Operating System and version (desktop or mobile): Linux Mint 19.2 (64)

```
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main  
atk                       2.36.0                        0    conda-forge
atk-1.0                   2.36.0               haf93ef1_0    conda-forge
boost-cpp                 1.73.0               h7b93d67_1    conda-forge
brotlipy                  0.7.0           py38h1e0a361_1000    conda-forge
bzip2                     1.0.8                h516909a_2    conda-forge
ca-certificates           2020.6.20            hecda079_0    conda-forge
cairo                     1.16.0            h3fc0475_1005    conda-forge
certifi                   2020.6.20        py38h32f6830_0    conda-forge
cffi                      1.14.0           py38hd463f26_0    conda-forge
chardet                   3.0.4           py38h32f6830_1006    conda-forge
cryptography              2.9.2            py38h766eaa4_0    conda-forge
curl                      7.71.1               he644dc0_0    conda-forge
expat                     2.2.9                he1b5a44_2    conda-forge
fontconfig                2.13.1            h1056068_1002    conda-forge
freetype                  2.10.2               he06d7ca_0    conda-forge
fribidi                   1.0.9                h516909a_0    conda-forge
future                    0.18.2           py38h32f6830_1    conda-forge
gdk-pixbuf                2.38.2               h3f25603_4    conda-forge
gettext                   0.19.8.1          hc5be6a0_1002    conda-forge
glib                      2.65.0               h6f030ca_0    conda-forge
gobject-introspection     1.64.1           py38h03d966d_1    conda-forge
graphite2                 1.3.13            he1b5a44_1001    conda-forge
gst-plugins-base          1.14.5               h0935bb2_2    conda-forge
gstreamer                 1.14.5               h36ae1b5_2    conda-forge
gtk2                      2.24.32              h194ddfc_3    conda-forge
harfbuzz                  2.4.0                hee91db6_5    conda-forge
hdf4                      4.2.13            hf30be14_1003    conda-forge
hdf5                      1.10.6          nompi_h3c11f04_100    conda-forge
icu                       67.1                 he1b5a44_0    conda-forge
idna                      2.10               pyh9f0ad1d_0    conda-forge
jpeg                      9d                   h516909a_0    conda-forge
jsoncpp                   1.8.4             hc9558a2_1002    conda-forge
krb5                      1.17.1               hfafb76e_1    conda-forge
ld_impl_linux-64          2.33.1               h53a641e_7  
libcurl                   7.71.1               hcdd3856_0    conda-forge
libedit                   3.1.20191231         h7b6447c_0  
libffi                    3.2.1             he1b5a44_1007    conda-forge
libgcc-ng                 9.1.0                hdf63c60_0  
libgfortran-ng            7.5.0                hdf63c60_6    conda-forge
libglu                    9.0.0             he1b5a44_1001    conda-forge
libiconv                  1.15              h516909a_1006    conda-forge
libnetcdf                 4.7.4           nompi_h84807e1_104    conda-forge
libpng                    1.6.37               hed695b0_1    conda-forge
libssh2                   1.9.0                hab1572f_2    conda-forge
libstdcxx-ng              9.1.0                hdf63c60_0  
libtiff                   4.1.0                hc7e4089_6    conda-forge
libuuid                   2.32.1            h14c3975_1000    conda-forge
libwebp-base              1.1.0                h516909a_3    conda-forge
libxcb                    1.13              h14c3975_1002    conda-forge
libxml2                   2.9.10               h72b56ed_1    conda-forge
lz4-c                     1.9.2                he1b5a44_1    conda-forge
ncurses                   6.2                  he6710b0_1  
netgen4smesh              6.2.1804             h6bb024c_0    trelau
occt                      7.4.0                h9121d39_5    conda-forge
openssl                   1.1.1g               h516909a_0    conda-forge
pango                     1.42.4               h7062337_4    conda-forge
pathlib2                  2.3.5            py38h32f6830_1    conda-forge
pcre                      8.44                 he1b5a44_0    conda-forge
pip                       20.1.1                   py38_1  
pixman                    0.38.0            h516909a_1003    conda-forge
pthread-stubs             0.4               h14c3975_1001    conda-forge
pycparser                 2.20               pyh9f0ad1d_2    conda-forge
pyocct                    7.4.0.0                  py38_1    trelau
pyopenssl                 19.1.0                     py_1    conda-forge
pypubsub                  4.0.3                      py_0    conda-forge
pysocks                   1.7.1            py38h32f6830_1    conda-forge
python                    3.8.3           cpython_he5300dc_0    conda-forge
python_abi                3.8                      1_cp38    conda-forge
readline                  8.0                  h7b6447c_0  
requests                  2.24.0             pyh9f0ad1d_0    conda-forge
setuptools                47.3.1                   py38_0  
six                       1.15.0             pyh9f0ad1d_0    conda-forge
smesh4pyocct              8.3.0.4              hf484d3e_0    trelau
sqlite                    3.32.3               h62c20be_0  
tbb                       2020.1               hc9558a2_0    conda-forge
tk                        8.6.10               hbc83047_0  
urllib3                   1.25.9                     py_0    conda-forge
vtk                       8.2.0           py38hf2e56f5_218    conda-forge
wheel                     0.34.2                   py38_0  
wxpython                  4.0.7.post2      py38hac52f33_3    conda-forge
xorg-kbproto              1.0.7             h14c3975_1002    conda-forge
xorg-libice               1.0.10               h516909a_0    conda-forge
xorg-libsm                1.2.3             h84519dc_1000    conda-forge
xorg-libx11               1.6.9                h516909a_0    conda-forge
xorg-libxau               1.0.9                h14c3975_0    conda-forge
xorg-libxdmcp             1.1.3                h516909a_0    conda-forge
xorg-libxext              1.3.4                h516909a_0    conda-forge
xorg-libxrender           0.9.10            h516909a_1002    conda-forge
xorg-libxt                1.2.0                h516909a_0    conda-forge
xorg-renderproto          0.11.1            h14c3975_1002    conda-forge
xorg-xextproto            7.3.0             h14c3975_1002    conda-forge
xorg-xproto               7.0.31            h14c3975_1007    conda-forge
xz                        5.2.5                h7b6447c_0  
zlib                      1.2.11               h7b6447c_3  
zstd                      1.4.4                h6597ccf_3    conda-forge

```

frmdstryr, 2020-07-08:
Are you using Wayland? If so you need to run  `export QT_QPA_PLATFORM=xcb` then the example.

mnesarco, 2020-07-08:
@frmdstryr I am not using wayland. I use Linux Mint (Cinnamon/xorg)

frmdstryr, 2020-07-08:
Ok, it looks like the same issue as https://github.com/trelau/pyOCCT/issues/32 (not sure what happened with that one). 

I'd recommend giving the Qt Viewer a try (just install PyQt5 and change the example use `from OCCT.Visualization.QtViewer import ViewerQt` instead.

mnesarco, 2020-07-08:
@frmdstryr 
> Ok, it looks like the same issue as #32 (not sure what happened with that one).
> 
> I'd recommend giving the Qt Viewer a try (just install PyQt5 and change the example use `from OCCT.Visualization.QtViewer import ViewerQt` instead.

Different error now.

Modified Example code i am trying to run:

https://github.com/trelau/pyOCCT/blob/master/examples/bop_assys.py

```
import time

from OCCT.BOPAlgo import BOPAlgo_Options
from OCCT.BRepAlgoAPI import BRepAlgoAPI_Fuse
from OCCT.TopTools import TopTools_ListOfShape

from OCCT.Exchange import ExchangeBasic
from OCCT.Visualization.QtViewer import ViewerQt  # Changed as suggested

fn = './models/wing_assy.brep'
wing_assy = ExchangeBasic.read_brep(fn)

fn = './models/fuse_assy.brep'
fuse_assy = ExchangeBasic.read_brep(fn)

BOPAlgo_Options.SetParallelMode_(True)
bop = BRepAlgoAPI_Fuse()
args = TopTools_ListOfShape()
args.Append(wing_assy)
bop.SetArguments(args)
tools = TopTools_ListOfShape()
tools.Append(fuse_assy)
bop.SetTools(tools)
print('Starting fuse...')
start = time.time()
bop.Build()
print('Complete in ', time.time() - start, ' seconds.')

v = ViewerQt()  # Changed as suggested
v.add(bop.Shape())
v.start()

```

Error:

```
(pyocct) mnesarco@devel03:examples$ python bop_assys.py 
Starting fuse...
Complete in  21.40282702445984  seconds.
Traceback (most recent call last):
  File "bop_assys.py", line 48, in <module>
    v = ViewerQt()
  File "/home/mnesarco/miniconda3/envs/pyocct/lib/python3.8/site-packages/OCCT/Visualization/QtViewer.py", line 496, in __init__
    self._the_view = QOpenCascadeWidget()
  File "/home/mnesarco/miniconda3/envs/pyocct/lib/python3.8/site-packages/OCCT/Visualization/QtViewer.py", line 101, in __init__
    window = Xw_Window(self.display_connection, hwnd)
AttributeError: 'QOpenCascadeWidget' object has no attribute 'display_connection'

```

frmdstryr, 2020-07-08:
You'll need to make the change from #61. Looks like there was an accidental mistake in a recent refactoring that was not caught as it's primarily tested on windows.

mnesarco, 2020-07-08:
@frmdstryr 
> You'll need to make the change from #61. Looks like there was an accidental mistake in a recent refactoring that was not caught as it's primarily tested on windows.

It works... but two windows appear, one black an the other gray, the solid is shown only if I do click the gray window but keypress events seems to be sent to both windows:

[image]

The message in console is:

```
TKOpenGl | Type: Other | ID: 0 | Severity: Medium | Message:
  OpenGl_Window::CreateWindow: window Visual is incomplete: no depth buffer, no stencil buffer

```

mnesarco, 2020-07-14:
Hi @trelau ,
I have been working with the QtViewer and it seems that the solution to the black screen is so simple:

You forgot to pass the parent to the QOpenCascadeWidget, so it creates a new window.
```
self._the_view = QOpenCascadeWidget()
```

The simple fix:
```
self._the_view = QOpenCascadeWidget(self)
```

I have also changed QOpenGLWidget to QGLWidget due to deprecation of QOpenGLWidget.

Complete patch:

```
diff --git a/OCCT/Visualization/QtViewer.py b/OCCT/Visualization/QtViewer.py
index b1c0a9a..5c75adc 100644
--- a/OCCT/Visualization/QtViewer.py
+++ b/OCCT/Visualization/QtViewer.py
@@ -36,7 +36,8 @@ from OCCT.V3d import V3d_Viewer, V3d_TypeOfOrientation
 from OCCT.gp import gp_Pnt
 from qtpy import QtCore
 from qtpy.QtGui import QPalette, QIcon
-from qtpy.QtWidgets import QApplication, QMainWindow, QOpenGLWidget, QFrame, QVBoxLayout
+from qtpy.QtWidgets import QApplication, QMainWindow, QFrame, QVBoxLayout
+from qtpy.QtOpenGL import QGLWidget
 
 try:
     from OCCT.SMDSAbs import SMDSAbs_Node
@@ -51,7 +52,7 @@ except ImportError:
 __all__ = ['ViewerQt']
 
 
-class QOpenCascadeWidget(QOpenGLWidget):
+class QOpenCascadeWidget(QGLWidget):
     """
     View for displaying shapes.
 
@@ -493,7 +494,7 @@ class ViewerQt(QMainWindow):
         frame = QFrame(self)
         layout = QVBoxLayout(frame)
         layout.setContentsMargins(0, 0, 0, 0)
-        self._the_view = QOpenCascadeWidget()
+        self._the_view = QOpenCascadeWidget(self)
         layout.addWidget(self._the_view)
         self.setCentralWidget(frame)
         self.show()

```

frmdstryr, 2020-07-14:
Nice catch, can you create a pull request?

trelau, 2020-07-15:
Fixed by #62

closed 2020-07-15 by trelau

FelisCorax, 2020-08-09:
Unless I'm doing something very wrong (which is far from impossible) #61 and #62 aren't included in the current conda package.

Also, there should probably be something about using qtviewer on the readme so linux users don't have to dig around in closed issues to figure out how to get pyocct to run.

giraldeau, 2020-12-14:
> Unless I'm doing something very wrong (which is far from impossible) #61 and #62 aren't included in the current conda package.
>
> Also, there should probably be something about using qtviewer on the [...]

I'm having the same issue. It might be obvious for someone that knows anaconda, but I'm new to it. 

I tried to install PyQt5 using conda: 

```
conda install -c conda-forge pyqt5-sip
```

Ant then I get this error: `ModuleNotFoundError: No module named 'PySide'`. I tried to install PySide, but there is no package for Python 3.8:

```
conda search -c conda-forge pyside
...
pyside                         1.2.4          py36_8  conda-forge
```

Tried with pip install (will it screw my conda environment? let's do it anyway), and it also fails: 

```
pip install pyside
Collecting pyside
  Downloading PySide-1.2.4.tar.gz (9.3 MB)
     |████████████████████████████████| 9.3 MB 1.4 MB/s 
    ERROR: Command errored out with exit status 1:
     command: /home/francisg/miniconda3/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-bwqbpsld/pyside/setup.py'"'"'; __file__='"'"'/tmp/pip-install-bwqbpsld/pyside/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-uj7p_m8c
         cwd: /tmp/pip-install-bwqbpsld/pyside/
    Complete output (1 lines):
    only these python versions are supported: [(2, 6), (2, 7), (3, 2), (3, 3), (3, 4)]
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

But actually, there is PySide2. And PySide6. It looks like its needed, but replaces in some way pyqt? I don't know, I'm confused.

Qt is a large dependency and a bit tricky to setup, so it would be nice to have some info, maybe just a link to other existing doc.