trelau/pyOCCT #63: boost-cpp dependency issue

cfriedland5, 2020-07-15:
The environment resulting from `conda create -n myenv -c defaults -c conda-forge -c trelau pyocct` fails to import `OCCT.SMESH` because of a DLL error

Reversing the channel priorities like so `conda create -n myenv -c trelau -c conda-forge -c defaults pyocct` results in an environment that works as expected

After investigating, I've determined that this is due to the version of boost-cpp which is 1.67.0 in the former case and 1.73.0 in the later. Running the original command with a precise specification of boost-cpp like so `conda create -n myenv -c defaults -c conda-forge -c trelau pyocct boost-cpp=1.73.0=h89d28cc_2` also results in an environment that works as expected.

How difficult would it be to fix the dependency specifications so conda does not make broken environments?

This issue was observed on a 64-bit x64 windows machine

frmdstryr, 2020-07-17:
pyOCCT doesn't explicitly require boost-cpp but SMESH does. 

A min version could be added in https://github.com/trelau/SMESH/blob/master/ci/conda/meta.yaml

trelau, 2020-07-19:
I can recreate the issue when I use `-c defaults` as the first channel. If I just do:

`conda create -n test -c conda-forge -c trelau pyocct` or `conda create -n test -c trelau -c conda-forge pyocct`

I get boost-cpp 1.73.0. Is there some reason the `-c defaults` option is throwing it off? Is that necessary in your case? When I put `-c defaults` in front of either command I do get boost-cpp 1.67.0.

cfriedland5, 2020-07-19:
I was trying to prefer packages from anaconda main if available when I observed the error but using `-c defaults`  is not necessary for me. If smesh requires a particular version of boost then that should be in the requirement though, right?

closed 2021-05-29 by trelau