trelau/pyOCCT #64: SetRaytracingMode not available in OCCT.V3d.V3d_View

mnesarco, 2020-07-16:
Hi Friends,
How can I activate RaytracingMode? it is not available in OCCT.V3d.V3d_View.

frmdstryr, 2020-07-17:
See https://dev.opencascade.org/doc/refman/html/class_v3d___view.html#ac62c6d1d0bbc13715b9730b5e12b1e41

Also your graphics card needs to support it.

trelau, 2020-07-19:
@mnesarco can you point to a specific OCCT method? I can't find a reference to `SetRaytracingMode` in the OCCT docs, but maybe I just am not finding it for some reason. Maybe that is an old method? I see some references to it from OCCT 6-ish?

mnesarco, 2020-07-19:
@trelau 
> @mnesarco can you point to a specific OCCT method? I can't find a reference to `SetRaytracingMode` in the OCCT docs, but maybe I just am not finding it for some reason. Maybe that is an old method? I see some references to it from OCCT 6-ish?

I was looking for this after seeing this:
https://dev.opencascade.org/index.php?q=node/898

But is an old post, I don't know if this feature was removed or renamed.

frmdstryr, 2020-08-17:
> I don't know if this feature was removed or renamed.

It was removed and replaced with [ChangeRenderingParams](https://dev.opencascade.org/doc/refman/html/class_v3d___view.html#ac62c6d1d0bbc13715b9730b5e12b1e41). This is not an issue with pyOCCT.

mnesarco, 2020-08-17:
Thank you @frmdstryr ,

Just for the reference this is the way to activate Raytracing mode:

`view->ChangeRenderingParams().Method = Graphic3d_RenderingMode::Graphic3d_RM_RAYTRACING;`

Where view is a V3d_View.

closed 2020-08-17 by mnesarco