trelau/pyOCCT #67: Image_Pixmap().Data pointer?

cylopa, 2020-10-06:
I am trying to convert V3d_View contents into a QPixmap object.
Using V3d_View we can Dump contents of the View to an image file. Using ToPixMap we can dump them to an Image_PixMap object. But Image_Pixmap.Data() returns an int? There is also InitWrapper method that takes a pointer as int?

cylopa, 2020-10-07:
Could  Image_Pixmap.Data() return as _bytes_ object?

 So we can do something like this:
`
image_pixmap = Image_PixMap()
v3d_view.ToPixMap(image_pixmap, width, height)
data_bytes: bytes = image_pixmap.Data()  # Currently returns an int.

q_pixmap = QPixmap()
q_bytearray = QByteArray(data_bytes)
q_pixmap.loadFromData(q_bytearray)
`

cylopa, 2020-10-08:
Or could it return a memoryview object?

data_bytes: memoryview= image_pixmap.Data()

trelau, 2021-01-18:
This seems pretty low-level, what are you trying to achieve?