trelau/pyOCCT #71: Error running examples/import_step.py

giraldeau, 2020-12-14:
I'm following the `Getting started using conda` instructions. Running the example produce the following error. I included all the output, but the error is at the end and says `invalid X window`. The event loop seems running, but nothing is displayed.

```
$ python import_step.py 
/home/francisg/miniconda3/envs/pyocct/lib/python3.8/site-packages/OCCT/Visualization/WxViewer.py:46: RuntimeWarning: SMESH module was not found for visualization.
  warnings.warn(msg, RuntimeWarning)
Gtk-Message: 11:24:16.278: Failed to load module "overlay-scrollbar"
Gtk-Message: 11:24:16.281: Failed to load module "atk-bridge"
Gtk-Message: 11:24:16.285: Failed to load module "canberra-gtk-module"
Press "c" to continue...

(import_step.py:35929): Gdk-WARNING **: 11:24:24.941: gdkdrawable-x11.c:952 drawable is not a pixmap or window
Traceback (most recent call last):
  File "/home/francisg/miniconda3/envs/pyocct/lib/python3.8/site-packages/wx/core.py", line 3383, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "/home/francisg/miniconda3/envs/pyocct/lib/python3.8/site-packages/OCCT/Visualization/WxViewer.py", line 144, in _init_wind
    wind = Xw_Window(self.display_connect, hwnd)
RuntimeError: Xw_Window, given invalid X window
```

## Expected Behavior

The window should display. If such a crash occurs, the process should exit.

## Steps to Reproduce (for bugs)

1. install pyocct from conda
2. run the example import_step.py 

## Your Environment

* Version used: pyocct                    7.4.0.0                  py38_1    trelau
* Operating System and version (desktop or mobile): Ubuntu Linux 18.04 (desktop)

frmdstryr, 2020-12-14:
https://github.com/trelau/pyOCCT/issues/60

giraldeau, 2020-12-14:
This issue is a duplicate, closing.

closed 2020-12-14 by giraldeau