trelau/pyOCCT #72: How to call specific function signatures?

russelmann, 2020-12-16:
Do I understand correctly that pyocct keeps function signatures from C++ to avoid clashing? For example, this is the way this function is supposed to be called:
```python
bounds = ShapeAnalysis.GetFaceUVBounds_(face, 0, 0, 0, 0) #argument zeros are required to call a specific C++ function and are not used as actual inputs in python
```
In this case I have a trouble trying to call function [ShapeAnalysis_Edge.PCurve](https://old.opencascade.com/doc/occt-7.0.0/refman/html/class_shape_analysis___edge.html#a2f7161f1532af718b2490e3658d3b258) as its third argument needs to be a handle to Geom2d_Curve which is abstract.
How to correctly call this function in pyocct?

frmdstryr, 2020-12-16:
Are you getting an error with your code? 

pyOCCT uses 7.4, the docs you have are from 7.0.  The function doesn't use a curve argument anymore https://dev.opencascade.org/doc/refman/html/class_shape_analysis.html#ad4f6532a707a510d838b2f73e23c6791

```python
face
Out[17]: <OCCT.TopoDS.TopoDS_Face at 0x7fde4c3344f0>

ShapeAnalysis.GetFaceUVBounds_(face, 0, 0, 0, 0)
Out[18]: (0.0, 25.4, -609.5999999999999, 0.0)
```

russelmann, 2020-12-16:
I gave an example of GetFaceUVBounds_ as a working code, and the following error occurs when I am trying to call [ShapeAnalysis_Edge.PCurve](https://old.opencascade.com/doc/occt-7.4.0/refman/html/class_shape_analysis___edge.html#a8a158d825bdf91371df49bddd086ec79) (fixed link to 7.4.0):
```
TypeError: PCurve(): incompatible function arguments. The following argument types are supported:
    1. (self: OCCT.ShapeAnalysis.ShapeAnalysis_Edge, edge: OCCT.TopoDS.TopoDS_Edge, face: OCCT.TopoDS.TopoDS_Face, C2d: OCCT.Geom2d.Geom2d_Curve, cf: float, cl: float, orient: bool) -> Tuple[bool, float, float]
    2. (self: OCCT.ShapeAnalysis.ShapeAnalysis_Edge, edge: OCCT.TopoDS.TopoDS_Edge, surface: OCCT.Geom.Geom_Surface, location: OCCT.TopLoc.TopLoc_Location, C2d: OCCT.Geom2d.Geom2d_Curve, cf: float, cl: float, orient: bool) -> Tuple[bool, float, float]
Invoked with: <OCCT.TopoDS.TopoDS_Edge object at 0x000002A13819D9B0>, <OCCT.TopoDS.TopoDS_Face object at 0x000002A13811BB30>, None, 0, 0, False
```
I am calling
```python
ShapeAnalysis_Edge.PCurve(edge, face, None, 0, 0, False)
```
and it is unclear what should I pass instead of None (argument 3), because it is not possible to instantiate abstract class OCCT.Geom.Geom_Surface.
Sorry for the confusion.

frmdstryr, 2020-12-16:
Gotcha, sorry for the confusion...  Do you have any C++ code as a reference? 

There's probably some API method to get an empty curve, then you need to pass that and it will update the reference. Eg [BRep_Tool::CurveOnSurface](https://old.opencascade.com/doc/occt-7.4.0/refman/html/class_b_rep___tool.html#a388ef940a84967b7199f9ade1a8a8004)

russelmann, 2020-12-17:
I have been calling the function in a wrong way. It is not static as I thought, so it can be called as follows
```python
sae = ShapeAnalysis_Edge()
sae.PCurve(edge, face, None, 0, 0, False)
```
However I still don't see a way to extract the Geom2d_Curve that this function is supposed to return. As I wrote above, in pyocct this function only returns Tuple[bool, float, float].
P.S. I think this is not related to None that I pass as argument 3, the same output is obtained if argument 3 contains a Geom2d_Curve returned from BRep_Tool::CurveOnSurface.
Most importantly, output Tuple[bool, float, float] is mentioned in TypeError when PCurve is called with wrong arguments (e.g. here passing no arguments).
```
TypeError: PCurve(): incompatible function arguments. The following argument types are supported:
    1. (self: OCCT.ShapeAnalysis.ShapeAnalysis_Edge, edge: OCCT.TopoDS.TopoDS_Edge, face: OCCT.TopoDS.TopoDS_Face, C2d: OCCT.Geom2d.Geom2d_Curve, cf: float, cl: float, orient: bool) -> Tuple[bool, float, float]
    2. (self: OCCT.ShapeAnalysis.ShapeAnalysis_Edge, edge: OCCT.TopoDS.TopoDS_Edge, surface: OCCT.Geom.Geom_Surface, location: OCCT.TopLoc.TopLoc_Location, C2d: OCCT.Geom2d.Geom2d_Curve, cf: float, cl: float, orient: bool) -> Tuple[bool, float, float]
Invoked with: <OCCT.ShapeAnalysis.ShapeAnalysis_Edge object at 0x000001E4E08C28F0>
```

trelau, 2021-01-18:
Passing things between Python and C++ by reference still needs a bit of work. In this case, the floats are immutable so they remain in the function signature to make sure you can choose the right overload, but by default immutable types that are non-const are also returned in the function output in addition to the regular output.

In this case, it may be better to just return the Geom2d_Curve in the output as well, but that takes modifying the binding sources.

If you are trying to get the 2d curve, the function [BRep_Tool.CurveOnSurface_](https://old.opencascade.com/doc/occt-7.5.0/refman/html/class_b_rep___tool.html#a388ef940a84967b7199f9ade1a8a8004) should work.

trelau, 2021-01-18:
this section of the documentation describes the logic a bit https://pyocct.readthedocs.io/en/latest/dev.html#reference-arguments

trelau, 2021-01-18:
@russelmann if you have a specific function you are looking to call but can't, can you open a ticket just for that? For each issues of this type we'll likely need to adjust the binding generator and write unit tests.

Going to close this issue.

closed 2021-01-18 by trelau