trelau/pyOCCT #73: Add AIS_CustomObject that can be implemented in python

frmdstryr, 2021-01-14:
It'd be nice to be able to create a custom `AIS_InteractiveObject` that can be implemented in python.

## Expected Behavior

Something along the lines of:

```python
from OCCT.AIS import AIS_CustomObject

class MyAISObject(AIS_CustomObject):

    def Compute(self, prs_mgr, prs, mode):
         # draw stuff...

    def ComputeSelection(self, projector, prs):
         # etc...

    def HandleClick(self, owner):
        # etc..
```

## Current Behavior

Attempting to extend the `AIS_InteractiveObject` doesn't work since there's no constructor and extending some other object doesn't work either since there's no delegation to the python virtual fn.

## Possible Solution

Create a `AIS_CustomObject` class that delegates certain calls to python as described in https://pybind11.readthedocs.io/en/stable/advanced/classes.html#overriding-virtual-functions-in-python

Would be nice to add to the 7.5 release. 

I can try to implement it if you think it'll work. I'm not sure how much of a performance impact it'll have.

trelau, 2021-01-16:
Fixed in https://github.com/trelau/pyOCCT/pull/74 right? Can we close this?

frmdstryr, 2021-01-16:
Yep working fine for what I needed.

closed 2021-01-16 by trelau

jf---, 2021-01-18:
This is inspired! I reckon an example would be amazing to document this new super power?