trelau/pyOCCT #78: Can't call AIS_InteractiveContext::EntityOwners method, no SelectMgr_IndexedMapOfOwner type

russelmann, 2021-04-20:
I cannot find a way to call the method [AIS_InteractiveContext::EntityOwners](https://old.opencascade.com/doc/occt-7.4.0/refman/html/class_a_i_s___interactive_context.html#a8cb62d753221eabc4b04f53585c1085f). It requires an argument of type `SelectMgr_IndexedMapOfOwner` which is not available through pyocct.

Any idea for a workaround to get the list of `SelectMgr_EntityOwner` from `AIS_InteractiveContext` or any over way to programmatically select subshapes of an `AIS_Shape`?

trelau, 2021-05-01:
Do you happen to have an example script that reproduces the issue using AIS_InteractiveContext::EntityOwners. I think I can bind the missing type, but would like a reproducible example to make sure and then add to the test suite. Thanks!

russelmann, 2021-05-02:
I sketched the following code using my current anaconda installation of pyocct.
```python
from OCCT.AIS import AIS_Shape
from OCCT.BRepPrimAPI import BRepPrimAPI_MakeBox
from OCCT.SelectMgr import SelectMgr_EntityOwner
from OCCT.TopAbs import TopAbs_EDGE
from OCCT.Visualization.QtViewer import ViewerQt

edge_mode = AIS_Shape.SelectionMode_(TopAbs_EDGE)

box = BRepPrimAPI_MakeBox(1, 1, 1).Shape()

viewer = ViewerQt()
ais_context = viewer._the_view.my_context
ais_shape = viewer.display_shape(box)
ais_context.Activate(ais_shape, edge_mode)

map_of_owner = SelectMgr_IndexedMapOfOwner()  # not binded
owners = ais_context.EntityOwners(map_of_owner, ais_shape, edge_mode)
entyty_owner: SelectMgr_EntityOwner = map_of_owner.FindIndex(1)
entyty_owner.SetSelected(True)

viewer.start()
```

In my understanding this snippet should create a box and select one of its edges.

russelmann, 2021-07-30:
Is this example ok or having it without the viewer (or else) would be of any help?