trelau/pyOCCT #85: [Packaging] Mostly empty "setup.py" makes pyOCCT mostly impossible to package

leycec, 2021-06-29:
I'd *love* to pyOCCT for Gentoo Linux and suspect other Linux distros feel similarly. Sadly, doing so requires a working `setup.py` script. [The current version mostly appears to be an empty placeholder](https://github.com/trelau/pyOCCT/blob/master/setup.py); there's no reference to pyOCCT's pure-Python dependency stack or CMake-based binding generation process, for example:

```
from setuptools import setup

setup(
    name='OCCT',
    version='7.5.2.0',
    packages=['OCCT', 'OCCT.Exchange', 'OCCT.Visualization'],
    package_data={'OCCT': ['*.so', '*.pyd', '*.dll', 'Visualization/_resources/*']},
    author='Trevor Laughlin',
    description='Python bindings for OpenCASCADE via pybind11.',
    url='https://github.com/trelau/pyOCCT',
    license='LGPL v2.1',
    platforms=['Windows', 'Linux']
)
```

That's... not gonna cut it.

Thankfully, this feature request has nothing to do with `pip`, `poetry`, PyPI, PyPA, or the special hell that is Python packaging standards. It's just `setuptools` all the way down. Without a working `setuptools`-based script, packaging pyOCCT for various platforms is basically infeasible. That saddens me, 'cause pyOCCT otherwise looks great.

Of course, it's fantastic that pyOCCT has such phenomenal support for `conda`! You've done a great job there, @trelau. :+1:

trelau, 2021-06-29:
thanks for filing the issue. i'm not sure i fully understand the issue, but let me provide some context if it helps:

the setup.py, in it's current form, is meant to be run after you build locally. That is, you run CMake and build things first, then afterwards run setup.py to install the python package. This is outlined in the REAMDE. Conda packages are also available if those are applicable. I'm on Windows, and some contributors helped set up what is currently supported for Linux in the Github Actions workflows and conda builds. Maybe there is a way to add support for Gentoo Linux, I'm just not sure how to do it.

Is what you are saying is that you expect the setup.py file to handle the entire process, from generating the bindings to compiling sources? Is there an example setup.py that is built the way you imagine?

leycec, 2021-06-30:
**Ugh!** Profuse apologies for the confusion, which is all my fault. The [Canadian Heat Dome](https://globalnews.ca/news/7989048/heat-wave-western-canada-what-is-causing-it) must have fried my fragile synapses.

Right. So, I'm currently packaging both pyOCCT and [CadQuery's pywrap, a mandatory dependency of OCP, itself a mandatory dependency of CadQuery](https://github.com/CadQuery/pywrap). Since pywrap and OCP are both strongly inspired by pyOCCT, I expected pyOCCT's `setup.py` to vaguely resemble [pywrap's `setup.py`](https://github.com/CadQuery/pywrap/blob/master/setup.py):

```
from setuptools import setup

setup(
    name='pywrap',
    version="0.1dev",
    entry_points={'console_scripts': 'pywrap = bindgen.__main__:main'},
    packages=['bindgen'],
    include_package_data = True,
    install_requires=[
        'click',
        'logzero',
        'path',
        'clang',
        'cymbal',
        'toml',
        'pandas',
        'joblib',
        'tqdm',
        'jinja2==2.11.3',
        'toposort',
        'pyparsing',
        'pybind11',
        'schema'
    ]
)
```

Basically, it'd be great if pyOCCT's `setup.py` could provide:

* A similar `install_requires` list of all mandatory runtime dependencies.
* A similar `tests_requires` list of all mandatory test-time dependencies. 

Optionally, it'd be even greater if pyOCCT's `setup.py` could perform *all* of the requisite build- and install-time logic required for a working pyOCCT installation. Technically, we can work around not having that on Gentoo (e.g., by just manually invoking `cmake` from our packaging scripts as you helpfully suggest).

Pragmatically, having a fully working and self-insulated `setup.py` file is a mandatory prerequisite for hosting pyOCCT on PyPI and making pyOCCT installable with `pip` – which is probably the biggest blocker to widespread adoption by the Python community that you'll face. Of course, I acknowledge that refactoring your fully working `conda` workflow to a `setup.py` workflow would not be fun and thus will probably never happen. But, hey... a packaging guy can hope and pray.

Thanks again for all the excellent work. You're making OpenCASCADE usable in 2021. :+1:

leycec, 2021-06-30:
**...fascinating.** I'd assumed that [some or all of the conda-specific dependencies listed in `meta.yaml` (e.g., `pybind11`) were also runtime dependencies](https://github.com/trelau/pyOCCT/blob/master/ci/conda/meta.yaml), but I suppose it makes sense that they're only build-time.

Thanks again, @frmdstryr! That's one hot car, too. :racing_car:

leycec, 2021-07-02:
And... **I'm closing this.** In hindsight, this was all a gross misunderstanding on my part. :frowning: 

It's *very* impressive that you've managed to keep pyOCCT dependency-free at both runtime and test-time. While it woud also be great to integrate the `cmake`-based binding generation into the `setup.py` script, that's another issue entirely and technically unnecessary for packaging under Gentoo and Arch Linux.

Keep up all the phenomenal work! pyOCCT rocks the OpenCASCADE stack.

closed 2021-07-02 by leycec