trelau/pyOCCT #93: Setting referenced based class attributes

capfish, 2021-11-01:
## Expected Behavior

I am unable to set the ShapeFix_Wireframe mode to remove small edges via its attribute [ModeDropSmallEdges](https://dev.opencascade.org/doc/refman/html/class_shape_fix___wireframe.html#a23e1dfe4e89f1695d1bf7c6a3a46f2fa)  Relatedly, I am also unable to change any of the modes (i.e. FixSolidMode, FixFreeFaceMode, etc) in [ShapeFix_Shape](https://dev.opencascade.org/doc/refman/html/class_shape_fix___shape.html)
```
>>> from OCCT.Exchange.Basic import ExchangeBasic
>>> from OCCT.ShapeFix import ShapeFix_Wireframe
>>> my_part = ExchangeBasic.read_step('path/to/my/step.step')
>>> wireframe_fixer = ShapeFix_Wireframe(my_part)
>>> wireframe_fixer.ModeDropSmallEdges = True
>>> wireframe_fixer.ModeDropSmallEdges()
True
```

## Current Behavior

```
>>> from OCCT.Exchange.Basic import ExchangeBasic
>>> from OCCT.ShapeFix import ShapeFix_Wireframe
>>> my_part = ExchangeBasic.read_step('path/to/my/step.step')
>>> wireframe_fixer = ShapeFix_Wireframe(my_part)
>>> wireframe_fixer.ModeDropSmallEdges = True
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'OCCT.ShapeFix.ShapeFix_Wireframe' object attribute 'ModeDropSmallEdges' is read-only
>>> wireframe_fixer.ModeDropSmallEdges()
False
```

## Possible Solution

This seems to be because those attributes are passed by reference in the C++ libraries, I'm not strong in pybind-fu so I'm not sure what would have to be changed, but I'm willing to try if you point me in the right direction!

## Context

I have been having issues with the capping plane feature and I was pointed to these shape healing tools on the [OpenCascade forums](https://dev.opencascade.org/content/debug-clipping-plane-not-working-some-step-files)  They work alright, but it is clear that I have some small disconnected edges that need removed and I'm unable to set the attribute. 

## Your Environment

* Version used: 7.5.2
* Operating System and version (desktop or mobile): Windows 10, Debian Stretch

russelmann, 2022-02-08:
I guess you intend to use method [MergeSmallEdges](https://dev.opencascade.org/doc/refman/html/class_shape_fix___wireframe.html#a9afbf908477a09d44c3d9f8a32a33eee), note that it has argument `theModeDrop` which you can set true as a workaround.

capfish, 2022-02-09:
Oh interesting I completely missed that.  I was intending to run [FixSmallEdges](https://dev.opencascade.org/doc/refman/html/class_shape_fix___wireframe.html#a035b9dfa50a1a51369377c6322e009f2), do you know if running `MergeSmallEdges` with `theModeDrop=True` followed by `FixSmallEdges` with `ModeDropSmallEdges` set to `False` is equivalent to just running `FixSmallEdges` with `ModeDropSmallEdges` set to `False`?

russelmann, 2022-02-09:
Seems that the workaround won't help using method FixSmallEdges. In case you don't need to call it on a compound, you should be fine calling CheckSmallEdges and MergeSmallEdges instead. You can check the implementation of FixSmallEdges [here ](https://github.com/Open-Cascade-SAS/OCCT/blob/V7_5_0/src/ShapeFix/ShapeFix_Wireframe.cxx), line 408. Lines 421-462 show an involved specific behavior for compounds with a recursive call which is problematic. However if you use any other topology is should suffice to only reimplement lines 463-466 in Python.