trelau/pyOCCT #94: Support for OCCT 7.6.0

trelau, 2021-11-14:
https://dev.opencascade.org/content/open-cascade-technology-760-released

Krande, 2022-04-22:
Hey, any chance you have had time to look at adding support for OCCT 7.6.1 for pyOCCT? :) 

I recently started experimenting on adapting the SWIG bindings of pythonocc-core (which is what I've relied upon for a long time) to support v7.6.1, but I see that there is a lot of files that needs updating. So I would be interested in anything that could help simplify the binding process :)

I also noticed you mentioned to a rewrite of the python binder (pyOCCT_binder) in #95 that would help automate the binding process? Is the code you referred to what is currently located in the pyOCCT_binder repo? 

Best Regards
Kristoffer

trelau, 2022-04-23:
Hi @Krande . I was working on a new version of the binding generator and applying it to 7.6. Although, that was a at least a few months ago and since then I've been short on time to work on these projects. I'll go ahead and share the repo for the new binding generator, but it's still in development.

Otherwise, I'm surprised PythonOCC hasn't caught up yet to 7.6.1. I think there is also an automated binding generator for that project, is it just a matter of updating it or are there larger changes in OCCT itself in 7.6 that make SWIG difficult to use?

There is also this project that seems to be pretty active if you are interested in trying it out https://github.com/CadQuery/OCP.

trelau, 2022-04-23:
Here is the (very much still WIP) new binding generator https://github.com/trelau/pybinder

Krande, 2022-04-23:
Thank you for responding so quickly, @trelau!

Regarding `PythonOCC`. Yes, it uses a binding generator called pythonocc-generator . It seems that it requires some fine-tuning, so I have started the process of updating the bindings to OCCT 7.6 in the issue tracker here: https://github.com/tpaviot/pythonocc-generator/issues/84.

Regarding `OCP`. Yes, as far as I can tell OCP is a bit closer to updating their bindings to 7.6.1. They have some `-alpha` versions on Anaconda `https://anaconda.org/CadQuery/ocp/files`. I will give OCP a try and see if maybe if it fits with my existing code base :)

It would be great if you could share the updated repo for the new binding generator! That way we would have 3 alternatives techniques for compiling python bindings onto OCCT.

I have been working with oce/occt through pythonocc/gmsh/ifcopenshell/cadquery for many years (and I have created a python package that utilizes all of them called [adapy](https://github.com/Krande/adapy). Ideally I can get the pythonocc-generator to work as I rely on some of the pythonocc-core extensions (which I believe is not part of occt) like `ShapeTesselator`. 

I still have lots to learn about the binding process itself (and I'm a complete novice in c++). But I finally have some time to dig more into the python binding process around OCCT, so hopefully I will be able to make myself useful.

Best Regards
Kristoffer