trelau/pyOCCT #97: Failed to build due to unnamed enums

adriendelsalle, 2022-02-21:
Hi @trelau thanks for this project ;)

I didn't managed to build bindings of `occt=7.5.3` due to unnamed enums in `occt` headers.
I get a lot of similar errors:
```
// ENUM: 
py::enum_<(unnamed enum at /home/adrien/micromamba/envs/binder/include/opencascade/Aspect_VKeyFlags.hxx:21:1)>(mod, "(unnamed enum at /home/adrien/micromamba/envs/binder/include/opencascade/Aspect_VKeyFlags.hxx:21:1)", "Key modifier, for combining with general key from Aspect_VKey.")
	.value("Aspect_VKeyFlags_NONE", (unnamed enum at /home/adrien/micromamba/envs/binder/include/opencascade/Aspect_VKeyFlags.hxx:21:1)::Aspect_VKeyFlags_NONE)
	.value("Aspect_VKeyFlags_SHIFT", (unnamed enum at /home/adrien/micromamba/envs/binder/include/opencascade/Aspect_VKeyFlags.hxx:21:1)::Aspect_VKeyFlags_SHIFT)
	.value("Aspect_VKeyFlags_CTRL", (unnamed enum at /home/adrien/micromamba/envs/binder/include/opencascade/Aspect_VKeyFlags.hxx:21:1)::Aspect_VKeyFlags_CTRL)
	.value("Aspect_VKeyFlags_ALT", (unnamed enum at /home/adrien/micromamba/envs/binder/include/opencascade/Aspect_VKeyFlags.hxx:21:1)::Aspect_VKeyFlags_ALT)
	.value("Aspect_VKeyFlags_MENU", (unnamed enum at /home/adrien/micromamba/envs/binder/include/opencascade/Aspect_VKeyFlags.hxx:21:1)::Aspect_VKeyFlags_MENU)
	.value("Aspect_VKeyFlags_META", (unnamed enum at /home/adrien/micromamba/envs/binder/include/opencascade/Aspect_VKeyFlags.hxx:21:1)::Aspect_VKeyFlags_META)
	.value("Aspect_VKeyFlags_ALL", (unnamed enum at /home/adrien/micromamba/envs/binder/include/opencascade/Aspect_VKeyFlags.hxx:21:1)::Aspect_VKeyFlags_ALL)
	.export_values();
```
related to unnamed enum, here declared in `Aspect_VKeyFlags.hxx`
```
//! Key modifier, for combining with general key from Aspect_VKey.
typedef unsigned int Aspect_VKeyFlags;

//! Key modifier, for combining with general key from Aspect_VKey.
enum
{
  Aspect_VKeyFlags_NONE  = 0,
  // reserve first 8 bits to combine value with Aspect_VKey
  Aspect_VKeyFlags_SHIFT = 1 <<  8, //!< Aspect_VKey_Shift
  Aspect_VKeyFlags_CTRL  = 1 <<  9, //!< Aspect_VKey_Control
  Aspect_VKeyFlags_ALT   = 1 << 10, //!< Aspect_VKey_Alt
  Aspect_VKeyFlags_MENU  = 1 << 11, //!< Aspect_VKey_Menu
  Aspect_VKeyFlags_META  = 1 << 12, //!< Aspect_VKey_Meta

  Aspect_VKeyFlags_ALL = Aspect_VKeyFlags_SHIFT | Aspect_VKeyFlags_CTRL | Aspect_VKeyFlags_ALT | Aspect_VKeyFlags_MENU | Aspect_VKeyFlags_META
};
```
TBH I don't really understand what's the benefit of having some enums declared that way.. any thoughts?

My local setup is:
- Ubuntu 21.10
- env created using `micromamba -f binder/environment.yml` extended the following way to have compilers and gl libs installed
```
name: binder
channels:
  - conda-forge
dependencies:
  - python=3.9
  - clangdev
  - python-clang
  - occt=7.5.3
  - tbb-devel
  - cmake
  - make
  - gxx_linux-64
  - mesa-libgl-devel-cos6-x86_64
  - libxi-devel-cos6-x86_64
  - libxext-devel-cos6-x86_64
  - glfw
  - pybind11
```

Running the generator gave me quite a lot of errors, but apparently no critical one:
```
(binder) adrien@tower:~/Documents/codes/occt/pyOCCT$ python binder/run.py -c binder/config.txt -o src
----------------------------------------------------------------------------------------------------
pyOCCT Binder
----------------------------------------------------------------------------------------------------
Include directories:
Clang: /home/adrien/micromamba/envs/binder/lib/clang/13.0.1/include
        OpenCASCADE: /home/adrien/micromamba/envs/binder/include/opencascade
        VTK: /home/adrien/micromamba/envs/binder/include/vtk-9.0
        TBB: /home/adrien/micromamba/envs/binder/include

Generating all_includes.h file...

Writing binding output files to: /home/adrien/Documents/codes/occt/pyOCCT/src

Parsing headers...
----------------------
DIAGNOSTIC INFORMATION
----------------------
Complete with 568 issues with lower than 4 severity not shown.
----------------------
Traversing headers...
Sorting binders...
Building includes...
Building imports...
Checking circular imports...
Binding templates...
Patched file: bind_BVH_BaseBox
Patched file: bind_BOPTools_PairSelector
Binding...
Patched file: Standard
Patched file: Standard
Patched file: Standard
Patched file: Standard
Patched file: Standard
Patched file: Geom
Patched file: AIS
Patched file: IVtkTools
Patched file: IVtkTools
Done!
----------------------------------------------------------------------------------------------------
```
Is it normal to have so many issues?

I had then to patch the `CMakeLists.txt` file line 76 to replace `find_package(OpenGL REQUIRED GLX EGL)` with `find_package(OpenGL REQUIRED) `. Didn't managed to find some conda packages to make it work another way.

Last step to reproduce is to run `make` and I got those kind of errors.
Git blame on [OCCT repo](https://github.com/Open-Cascade-SAS/OCCT/blame/600ee856316c1a9eb340779b32de511b74465896/src/Aspect/Aspect_VKeyFlags.hxx) doesn't mention any recent change. Do you see what I'm missing here?

Thanks!

adriendelsalle, 2022-02-28:
It's apparently a change in the way `clang` is reporting the anonymous enums, related to the binder project.
See https://github.com/trelau/pyOCCT_binder/blob/50853bc763f53e965dc1ade8c75f5a4a017a07ca/pybinder/core.py#L2379

I'll push a PR to fix that!

closed 2022-02-28 by adriendelsalle

ThorNielsen-RD8, 2022-06-15:
Has this yet been fixed anywhere? I tried to build yesterday, but still get the same error as in the first comment of this issue.

ThorNielsen-RD8, 2022-06-15:
It appears the issue is deeper than just changing the spelling from `'(anonymous enum'` to `'(unnamed enum'`, since after replacing all instances of the first with the latter, the build proceeds until ~30%, where it then fails with errors about undeclared variables in BinMXCAFDoc.cxx, e.g. `pyOCCT/src/BinMXCAFDoc.cxx:246:81: error: ‘MaterialVersionMajor_1’ was not declared in this scope`.