trelau/pyOCCT #101: Are you willing to continue maintenance? Can't build normally.

ovo-Tim, 2024-01-28:
## Current Behavior
```
❯ cmake --build .
[  0%] Built target Adaptor2d
[  1%] Built target Adaptor3d
[  1%] Building CXX object CMakeFiles/AdvApp2Var.dir/src/AdvApp2Var.cxx.o
In file included from /tmp/download/pyOCCT-master/src/AdvApp2Var.cxx:63:
/tmp/download/pyOCCT-master/src/bind_NCollection_Sequence.hxx: In function ‘void bind_NCollection_Sequence(pybind11::module&, const std::string&, const pybind11::module_local&)’:
/tmp/download/pyOCCT-master/src/bind_NCollection_Sequence.hxx:61:85: Error：Cannot declare pointer to member 'void'
   61 | cls_NCollection_Sequence.def("Remove", (void (NCollection_Sequence<TheItemType>::*)(Iterator &)) &NCollection_Sequence<TheItemType>::Remove, "Remove one item", py::arg("thePosition"));
      |                                                                                     ^~~~~~~~
/tmp/download/pyOCCT-master/src/bind_NCollection_Sequence.hxx:61:41: Error：expected primary-expression before ‘void’
   61 | cls_NCollection_Sequence.def("Remove", (void (NCollection_Sequence<TheItemType>::*)(Iterator &)) &NCollection_Sequence<TheItemType>::Remove, "Remove one item", py::arg("thePosition"));
      |                                         ^~~~
/tmp/download/pyOCCT-master/src/bind_NCollection_Sequence.hxx:61:41: Error：expected ‘)’ before ‘void’
   61 | cls_NCollection_Sequence.def("Remove", (void (NCollection_Sequence<TheItemType>::*)(Iterator &)) &NCollection_Sequence<TheItemType>::Remove, "Remove one item", py::arg("thePosition"));
      |                                        ~^~~~
      |                                         )
/tmp/download/pyOCCT-master/src/bind_NCollection_Sequence.hxx:70:90: Error：Cannot declare pointer to member 'void'
   70 | cls_NCollection_Sequence.def("InsertAfter", (void (NCollection_Sequence<TheItemType>::*)(Iterator &, const TheItemType &)) &NCollection_Sequence<TheItemType>::InsertAfter, "InsertAfter the position of iterator", py::arg("thePosition"), py::arg("theItem"));
      |                                                                                          ^~~~~~~~
/tmp/download/pyOCCT-master/src/bind_NCollection_Sequence.hxx:70:46: Error：expected primary-expression before ‘void’
   70 | cls_NCollection_Sequence.def("InsertAfter", (void (NCollection_Sequence<TheItemType>::*)(Iterator &, const TheItemType &)) &NCollection_Sequence<TheItemType>::InsertAfter, "InsertAfter the position of iterator", py::arg("thePosition"), py::arg("theItem"));
      |                                              ^~~~
/tmp/download/pyOCCT-master/src/bind_NCollection_Sequence.hxx:70:46: Error：expected ‘)’ before ‘void’
   70 | cls_NCollection_Sequence.def("InsertAfter", (void (NCollection_Sequence<TheItemType>::*)(Iterator &, const TheItemType &)) &NCollection_Sequence<TheItemType>::InsertAfter, "InsertAfter the position of iterator", py::arg("thePosition"), py::arg("theItem"));
      |                                             ~^~~~
      |                                              )
make[2]: *** [CMakeFiles/AdvApp2Var.dir/build.make:76：CMakeFiles/AdvApp2Var.dir/src/AdvApp2Var.cxx.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:811：CMakeFiles/AdvApp2Var.dir/all] Error 2
make: *** [Makefile:136：all] Error 2
```

## Steps to Reproduce (for bugs)
Just follow the [steps that were written in readme.](https://github.com/trelau/pyOCCT?tab=readme-ov-file#building-from-sources)

## Your Environment

* Version used: 7.5.3.1
* Operating System and version (desktop or mobile): 
```
❯ neofetch
                   -`                    tim@TimTu-Arch 
                  .o+`                   -------------- 
                 `ooo/                   OS: Arch Linux x86_64 
                `+oooo:                  Host: Code01 Ver2.0 1 
               `+oooooo:                 Kernel: 6.7.1-arch1-1 
               -+oooooo+:                Uptime: 24 mins 
             `/:-:++oooo+:               Packages: 1338 (pacman) 
            `/++++/+++++++:              Shell: bash 5.2.26 
           `/++++++++++++++:             Resolution: 2560x1600 
          `/+++ooooooooooooo/`           DE: Plasma 5.27.10 
         ./ooosssso++osssssso+`          WM: kwin 
        .oossssso-````/ossssss+`         Theme: NephriteLight [Plasma], Breeze [GTK2/3] 
       -osssssso.      :ssssssso.        Icons: [Plasma], Colloid [GTK2/3] 
      :osssssss/        osssso+++.       Terminal: konsole 
     /ossssssss/        +ssssooo/-       Terminal Font: Hack [SRC] 10 
   `/ossssso+/:-        -:/+osssso+-     CPU: AMD Ryzen 7 6800H with Radeon Graphics (16) @ 4.785GHz 
  `+sso+:-`                 `.-/+oso:    GPU: AMD ATI Radeon 680M 
 `++:.                           `-/+/   Memory: 11474MiB / 30808MiB 
 .`                                 `/
❯ conda list
# packages in environment at /tmp/occt_env:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
aom                       3.6.1                h59595ed_0    conda-forge
binutils_impl_linux-64    2.40                 hf600244_0    conda-forge
bzip2                     1.0.8                hd590300_5    conda-forge
[... 124 lines cut ...]
xorg-damageproto          1.2.1             h7f98852_1002    conda-forge
xorg-fixesproto           5.0               h7f98852_1002    conda-forge
xorg-glproto              1.4.17            h7f98852_1002    conda-forge
xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
xorg-libice               1.1.1                hd590300_0    conda-forge
xorg-libsm                1.2.4                h7391055_0    conda-forge
xorg-libx11               1.8.7                h8ee46fc_0    conda-forge
xorg-libxau               1.0.11               hd590300_0    conda-forge
xorg-libxdamage           1.1.5                h7f98852_1    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h0b41bf4_2    conda-forge
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxrandr            1.5.2                h7f98852_1    conda-forge
xorg-libxrender           0.9.11               hd590300_0    conda-forge
xorg-libxt                1.3.0                hd590300_1    conda-forge
xorg-randrproto           1.5.0             h7f98852_1001    conda-forge
xorg-renderproto          0.11.1            h7f98852_1002    conda-forge
xorg-util-macros          1.19.3               h7f98852_0    conda-forge
xorg-xextproto            7.3.0             h0b41bf4_1003    conda-forge
xorg-xf86vidmodeproto     2.3.1             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
zlib                      1.2.13               hd590300_5    conda-forge
zstd                      1.5.5                hfc55251_0    conda-forge
```

trelau, 2024-01-28:
hi @ovo-Tim . although i wish i could, it's been difficult to find much time to work on this project, so i can't commit to continued maintenance or any levels of support at this time.

If you haven't already, check out [pythonocc](https://github.com/tpaviot/pythonocc-core). It has a pretty stable build system and I think is readily available on conda. if you're wondering why i made pyOCCT, i answered that [here](https://stackoverflow.com/questions/62783383/pyocct-vs-pythonocc-for-new-project-2020) on stackoverflow just fyi

ovo-Tim, 2024-01-29:
> If you haven't already, check out [pythonocc](https://github.com/tpaviot/pythonocc-core). It has a pretty stable build system and I think is readily available on conda.
@trelau Thanks for your reply. In fact, I have used pythonocc for a long time. But I'm trying to switch to pyOCCT. Compared with pythonocc, I think pyOCCT provide interface that closer to native and support more module.

> although i wish i could, it's been difficult to find much time to work on this project, so i can't commit to continued maintenance or any levels of support at this time.
If I made some updates, should I create a pull request?

frmdstryr, 2024-01-29:
If you're on linux you can try https://github.com/frmdstryr/pyOCCT/tree/v7.8.0

ovo-Tim, 2024-01-30:
@frmdstryr Thanks a lot.

ovo-Tim, 2024-01-30:
@frmdstryr I have built successfully!

ovo-Tim, 2024-01-30:
I have built a new conda package. https://anaconda.org/ovo-tim/pyocct

ovo-Tim, 2024-01-30:
@frmdstryr Hi, I found that the conda_build can't output the right package. 
[image]
https://anaconda.org/ovo-tim/pyocct/7.8.0.0/download/linux-64/pyocct-7.8.0.0-py39_100.tar.bz2

ovo-Tim, 2024-01-30:
> @frmdstryr Hi, I found that the conda_build can't output the right package. (It outputted wrong python version) ![image](https://private-user-images.githubusercontent.com/116942368/300726119-83ef8416-2860-49fd-b622-aed578b8eda8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY2MDI4MDcsIm5iZiI6MTcwNjYwMjUwNywicGF0aCI6Ii8xMTY5NDIzNjgvMzAwNzI2MTE5LTgzZWY4NDE2LTI4NjAtNDlmZC1iNjIyLWFlZDU3OGI4ZWRhOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEzMFQwODE1MDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02YTA0NTE1YzAzMzEzMGU2ZjY1MDFhOTRkODdhZTUxOTFhNzNlYjExNGYwMTEwMDk4NzI3MTkxM2ExNDc4ZjZlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.JFyNnFLwp_x3s_jwUT9Ni2MfAmfKjE6Csbg1X9HwSyk) https://anaconda.org/ovo-tim/pyocct/7.8.0.0/download/linux-64/pyocct-7.8.0.0-py39_100.tar.bz2

I think we should set python ver at 'meta.yaml', like:
``` yaml
  host:
    - python {{ python }}
    - pybind11
    - occt ==7.8.0
    - tbb-devel

  run:
    - python {{ python }}.*
    - occt ==7.8.0
```

ovo-Tim, 2024-01-30:
https://anaconda.org/ovo-tim/pyocct/files
All right, pyOCCT for linux is available now. But I can't make windows and MacOS builder work.

ovo-Tim, 2024-01-30:
@frmdstryr How do you think if we create a new repository called pyOCCT2, and continue to maintain it.😉

ovo-Tim, 2024-01-31:
@frmdstryr Hi, I found that I couldn't run pyOCCT-7.8.0:
```
❯ python import_step.py
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 674, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 571, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1176, in create_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
ImportError: generic_type: type "TColgp_Array2OfPnt" referenced unknown base type "NCollection_Array1<gp_Pnt>"

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 674, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 571, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1176, in create_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
ImportError: initialization failed

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/tmp/下载/pyOCCT-7.8.0/examples/import_step.py", line 20, in <module>
    from OCCT.Graphic3d import Graphic3d_NOM_ALUMINIUM
ImportError: initialization failed
``` 

By the way, can you enable issues in your repository？

frmdstryr, 2024-01-31:
Oh sorry, it looks like I forgot to commit a few changes I had locally. I updated the branch and binder repo.

I don't want to create a fork or separate issue tracker. Perhaps someone can contribute updates to fix CI & windows builds.

ovo-Tim, 2024-01-31:
@frmdstryr Hi, I found a strange bug. You have changed `pyOCCT_binder`, but you didn't update the code in https://github.com/frmdstryr/pyOCCT_binder. That makes this repository disordered.

frmdstryr, 2024-01-31:
I guess committing a submodule doesn't update the branch. Anyways I merged the binder branch I was on to my master branch, does it work now?

ovo-Tim, 2024-01-31:
> I guess committing a submodule doesn't update the branch. Anyways I merged the binder branch I was on to my master branch, does it work now?

Thanks for your work. But I think I'm in a big trouble now. It seems the link was linked to an inexistence branch.
[image]
[image]

ovo-Tim, 2024-02-01:
@frmdstryr I don't know why, but it still happened on the c[onda package](https://anaconda.org/ovo-tim/pyocct) that I built. 

If I try to build locally, it outputs:
```
❯ python import_step.py
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 666, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 565, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1173, in create_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
ImportError: generic_type: type "NCollection_BaseAllocator" referenced unknown base type "Standard_Transient"

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 666, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 565, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1173, in create_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
ImportError: initialization failed

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 666, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 565, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1173, in create_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
ImportError: initialization failed
```

frmdstryr, 2024-02-01:
I use some custom occt patches so my occt and pyocct are always built with the exact same compiler version. Can you try building occt locally and then pyocct to make sure there is no difference in compiler versions?

ovo-Tim, 2024-02-01:
@frmdstryr Thanks for your help again, I will try that tomorrow. (I hope you don't mind me asking, but may I know which country you are from? This way, I can better coordinate our communication considering the time zones. I'm based in China, and it's quite late for me to continue working now.😉)

frmdstryr, 2024-02-02:
I'm in the eastern timezone. I got CI to actually build and test but it appears the bindings are not being generated correctly for some reason (it is not doing some +before_type lines that should fix the errors).

ovo-Tim, 2024-02-02:
I'm trying to make CI build the latest VTK and OCCT. I hope that can help.

frmdstryr, 2024-02-02:
Since the bindings are not correct, I think rebuilding occt is not necessary (not sure what I was thinking the other day).

ovo-Tim, 2024-02-02:
Look forward to your work. :wink:

frmdstryr, 2024-02-02:
Looks like the `before_type` lines in the config didn't work with typedefs (I guess I had added them manually and did not regenerate since when I build locally regenerating flushes the build cache). I pushed an update so hopefully that fixes it.

ovo-Tim, 2024-02-03:
@frmdstryr It looks like the error message has changed, but it still doesn't work.
```
❯ python import_step.py
ImportError: generic_type: type "StepGeom_Array2OfSurfacePatch" referenced unknown base type "NCollection_Array1<opencascade::handle<StepGeom_SurfacePatch> >"

The above exception was the direct cause of the following exception:

ImportError: initialization failed
```

ovo-Tim, 2024-02-03:
I'm trying to build OCCT 7.8.0 (so that I can build pyOCCT locally). But I met a problem: https://dev.opencascade.org/content/build-failed-vtk-addlibrary-cannot-create-target-tkernel-because-imported-target-same-name

ovo-Tim, 2024-02-03:
Hi. Recently, I have read a [post](https://stackoverflow.com/a/67895367/20847787), it said OCP is better than pyOCCT and pythonOCC. Have you ever heard of [OCP](https://github.com/CadQuery/OCP). Is it more convenient to maintenance? I found it hasn't been upgraded to the latest version.

trelau, 2024-02-03:
> Hi. Recently, I have read a [post](https://stackoverflow.com/a/67895367/20847787), it said OCP is better than pyOCCT and pythonOCC. Have you ever heard of [OCP](https://github.com/CadQuery/OCP)? Is it more convenient to maintenance? I found it hasn't been upgraded to the latest version.

as far as i know, OCP uses pybind11 to generate python bindings, clang to parse the headers, and some sort of templating system to actually write the pybind11 bindings. so, i think, very similar to pyOCCT except I use clang to parse the headers and a "hand rolled" binding generator written in python. in my view, the destination is the same (python bindings of OCCT via pybind11), but the mechanisms to get there are different. if OCP is somehow easier to keep up to date and does a better job of handling all of the OCCT C++ nuances, then it might be worth a look, i guess.

frmdstryr, 2024-02-03:
@trelau, how did the isinstance checks work on the HArray's before?  I had to disable that check https://github.com/frmdstryr/pyOCCT/commit/a143524e4c626fa284cdeb02ff7c31bae5757237 to get the linux builds to pass the tests.

IDK how it ever worked since the second base class is missing in the `bind_Define_HArrayx.cxx` headers?

trelau, 2024-02-03:
@frmdstryr i don't remember the details, but looks like I had to do some special tricks [here](https://github.com/trelau/pyOCCT/blob/master/inc/bind_Define_HArray1.hxx#L51) if that gives you any clues. Looks like I added methods to the type from the base class but didn't have the inheritance for some reason. why the isinstance worked (or didn't) i can't think of off the top of my head...

closed 2024-06-23 by ovo-Tim